RPi-Monitor/RPi-Monitor #213: Installation links are broken

JohanSF, 2018-03-25:
https://xavierberger.github.io/RPi-Monitor-docs/11_installation.html

http://goo.gl/vewCLL is broken, so I can't install.

XavierBerger, 2018-03-25:
I just click on you link and I've got exactly what I was expecting. 
```
# RPi-Monitor official repository
deb http://giteduberger.fr rpimonitor/
```
I just retest the procedure right now and everything looks right...
I think your issue is somewhere else...

closed 2018-03-25 by XavierBerger

JohanSF, 2018-03-25:
http://goo.gl/vewCLL gives me "This site can’t be reached, goo.gl refused to connect."

XavierBerger, 2018-03-25:
The problem is located into your network.
Copy the content of the file shown upper into the file `/etc/apt/sources.list.d/rpimonitor.list` and continue the procedure. This should fix your issue.