RPi-Monitor/RPi-Monitor #215: Undefined value on correct voltage reading

x2k13, 2018-04-04:
Finally my Pi 3 B+ arrived and so I thought I'd make a fresh start with my media center. 
Running latest OSMC version (2018.03-2):
[https://osmc.tv/2018/03/osmcs-march-update-is-here-with-pi-3-b-support/](url)

After installing rpimonitor I noticed under the "Voltage" entry there's just a "undefined" value.
[image]
the ./cpu.conf  is untouched:
```
dynamic.2.name=cpu_voltage
dynamic.2.source=vcgencmd measure_volts core
dynamic.2.regexp=(\d+.\d+)V
dynamic.2.postprocess=
dynamic.2.rrd=
```
When I issue the command on console, I get the correct value:
```
vcgencmd measure_volts core
volt=1.2000V
```
I have no idea what's wrong with that, I already tried several regex variants but to no avail.

avma, 2018-04-06:
Hi,
I reinstalled  rpimonitor_latest.deb on a rpi B and rpi2 b. Temperature is not showing on both, temperature.conf was untouched...

[image]

My regards
Avi

x2k13, 2018-05-20:
Meanwhile, OSMC got an update to version 2018.04-2, but the issue still persists.
I still can't get RPi-Monitor to display voltage values.

Leepic, 2019-01-04:
@x2k13 can you give me the whole config file?

x2k13, 2019-01-17:
@Leepic
Sorry, there hasn't been any response to this issue by the Dev in months. Meanwhile I switched to NetData and ditched all Rpi-Monitor files. Can't help with that anymore, so I'm closing this now.

closed 2019-01-17 by x2k13