RPi-Monitor/RPi-Monitor #217: rpi mon v 2.13 beta 6

avma, 2018-04-06:
Hi there,
 I removed and reinstalled rpi-monitore to the latest beta 6 version, on  rpi b, rpi2 b and rpi 3 b all running on  a headless Raspbian Stretch and latest kernels.
rpi b and rpi3 b does nor recognize statement in the format: "  web.status.1.content.1.**title="WiFi"** "
but does " web.statistics.1.content.9.**name=WiFi** "
[image]
This is probably on the system level (Apache 2.4.35, php7....) any  ideas?
Thanks
Avi

XavierBerger, 2018-04-11:
Yes, this is a change introduce in v2.13... if you install v2.12 and read currently "work in progress" documentation, you may experience such issue.
I'll keep this issue open until publication.

patricktokeeffe, 2018-11-10:
I'm seeing similar behavior in `2.12-r0`. Using "title" results in "undefined" displayed:
```
web.statistics.1.content.1.title="Time sync"
web.statistics.1.content.1.graph.1=ntp_sources
web.statistics.1.content.1.ds_graph_options.ntp_sources.label=Num sources
```
[image]

But "name=" works (sort of - the quotation marks are interpreted as part of the name):
```
web.statistics.1.content.1.name="Time sync"
web.statistics.1.content.1.graph.1=ntp_sources
web.statistics.1.content.1.ds_graph_options.ntp_sources.label=Num sources
```
[image]

---
```
lar@rv-server:~$ apt-cache policy rpimonitor
rpimonitor:
  Installed: 2.12-r0
  Candidate: 2.12-r0
  Version table:
 *** 2.12-r0 500
        500 http://giteduberger.fr rpimonitor/ Packages
        100 /var/lib/dpkg/status
```

heitorPB, 2019-04-10:
Any news on the version 2.13, @XavierBerger ?

closed 2020-06-01 by XavierBerger