RPi-Monitor/RPi-Monitor #218: "Possible SYN flooding on port 8889. Sending cookies" message on Odroid-HC1

houdini69, 2018-04-11:
rpi-monitor is running on port 8889 and I have got a lot of this message in dmesg:
TCP: request_sock_TCP: Possible SYN flooding on port 8889. Sending cookies.  Check SNMP counters.
How to avoid that?

Yury-MonZon, 2018-04-30:
Same for me. I'm absolutely sure that no one is flooding in my local network.

dideldumthecat, 2018-08-08:
I can confirm that. Mine is running on port 8888 and I get the same message. I have to reload all pages multiple times before they load correctly (start page, statistics page).

This is very annoying- any ideas on how to fix that?

Leepic, 2019-01-02:
I think I have find the problem with some tests on my RaspberryPi 3B and RaspberryPi B+ with RPi-Monitor version 2-12-r0. It's the same for the latest version (2-13).

The perl script located at _/usr/bin/rpimonitord_ [line 575](https://github.com/XavierBerger/RPi-Monitor/blob/f24416365777de403ff16248cd8cda0a41721424/src/usr/bin/rpimonitord#L574) doesn't contains the Listen argument that define the queue size. By default it's 5 regarding to the [CPAN Perl documentation](https://metacpan.org/pod/IO::Socket::INET), it's about the IO::Socket::INET module but RPi-Monitor use HTTP::Daemon that depend on it.

## Correction ##

(RPi-Monitor version 2-12-r0)

```shell
vi +555 /usr/sbin/rpimonitord
```

- add the line below

```perl
Listen => SOMAXCONN,
```

after

```perl
this->{'server'} = new HTTP::Daemon ( ReuseAddr => 1,
```

- Restart RPi-Monitor

systemctl restart rpimonitor.service

## Before ##

```shell
ss -nl '( sport = :8888 )'
Netid  State      Recv-Q Send-Q                                                 Local Address:Port                                                   Peer Address:Port 
tcp    LISTEN     0      5                                                                  *:8888                                                              *:*
```

## After ##

```shell
ss -nl '( sport = :8888 )'
Netid  State      Recv-Q Send-Q                                                 Local Address:Port                                                   Peer Address:Port 
tcp    LISTEN     0      128                                                                *:8888                                                              *:*     
```

Now you can test the webUI and see if the error still occur:

```shell
dmesg -T | grep request_sock_TCP
[mer. janv.  2 21:52:04 2019] TCP: request_sock_TCP: Possible SYN flooding on port 8888. Sending cookies.  Check SNMP counters.
```

If you want to known what's the maximum number of the queue size on your system, you can type:

```perl
perl -MSocket -le 'print SOMAXCONN' 
128
```

Another solution is to run RPi-Monitor behind a reverse proxy like Apache or NginX (and it's better for the security as the [documentation say](https://xavierberger.github.io/RPi-Monitor-docs/34_autentication.html#authentication-and-secure-access)). Apache doesn't have this problem because by default the queue length is 128:

```shell
ss -nl '( sport = :80 )'
Netid  State      Recv-Q Send-Q                                                 Local Address:Port                                                   Peer Address:Port 
tcp    LISTEN     0      128                                                                *:80                                                                *:*
```

@dideldumthecat @Yury-MonZon and @houdini69 can you confirm me if this correct the issue?

dideldumthecat, 2019-01-03:
@Leepic : Great stuff, I can confirm that your fix works for me with version `2.12-r0` on Raspbian! Now all pages load and reload without errors. Thanks! 

On a side note, I had to  use `sudo vi +555 /usr/bin/rpimonitord`, as the file was located in `/usr/bin` rather than `/usr/sbin`.

Leepic, 2019-01-03:
@dideldumthecat thank you for your answer. I updated my previous comment following your note.
I will submit a pull request soon if all is okay :-).

Leepic, 2019-01-03:
Here the pull request: #247

Leepic, 2019-07-20:
@houdini69 this issue is resolved in the develop branch, so can you close this issue? Thanks.

slowrunner, 2019-09-16:
I keep see this issue every day or two - Possible SYN flooding on port 8888, on Raspbian Stretch on Raspberry Pi 3B:
Linux Carl 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

With respect to:
> sudo vi +555 /usr/bin/rpimonitord

Is it ok to edit with nano? 
```
sudo nano /usr/bin/rpimonitord
```

(What is the +555 for in the vi command?)

SuperSandro2000, 2019-09-17:
> Is it ok to edit with nano?

Doesn't matter.

>What is the +555 for in the vi command?

Vi jumps then straight to line 555.