RPi-Monitor/RPi-Monitor #219: Service active but doesn't remote accessible

kauanmocelin, 2018-04-12:
Rpi-monitor service is accessible but after a while it is unavailable.

My logs:
● rpimonitor.service - LSB: RPi-Monitor daemon
   Loaded: loaded (/etc/init.d/rpimonitor; generated; vendor preset: enabled)                                                                                                            
   Active: active (running) since Wed 2018-04-11 20:10:23 -03; 15h ago
     Docs: man:systemd-sysv-generator(8)                                                                                                                                                 
  Process: 12452 ExecStop=/etc/init.d/rpimonitor stop (code=exited, status=0/SUCCESS)                                                                                                    
  Process: 12469 ExecStart=/etc/init.d/rpimonitor start (code=exited, status=0/SUCCESS)                                                                                                  
   CGroup: /system.slice/rpimonitor.service                                                                                                                                              
                                                                                                                                                                                         
Apr 11 20:10:22 raspi systemd[1]: Starting LSB: RPi-Monitor daemon...                                                                                                                    
Apr 11 20:10:23 raspi rpimonitor[12469]: Starting RPi-Monitor:.                                                                                                                          
Apr 11 20:10:23 raspi systemd[1]: Started LSB: RPi-Monitor daemon.

According to the logs service is active - running but i can't acess remotely by firefox.

Leepic, 2019-01-04:
In your first command result `tcp 6 0 0.0.0.0:8888 0.0.0.0:* LISTEN -` there is no program associated. In my case I have this:

`tcp        0      0 0.0.0.0:8888            0.0.0.0:*               LISTEN      15607/perl      `

What's the output of those commands:

`sudo lsof -i :8888`
`ps -ef | grep perl`

Leepic, 2019-07-14:
@kauanmocelin : any update on your issue?

kauanmocelin, 2019-07-15:
I don't use this service anymore, thanks.

Leepic, 2019-07-20:
@kauanmocelin No problem, so can you close this issue, please? Thanks.

closed 2019-07-22 by kauanmocelin