RPi-Monitor/RPi-Monitor #224: How to configure for wifi usage on Rpi3

DdraigX, 2018-05-19:
I was wondering if someone could give me a quick run down on how to configure this to monitor wifi. 

I'm guessing it is using the default config for a Rp3, because my interfaces config is empty. 

My /etc/network/interfaces looks like the following:

`# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d
`

So I tried to edit the file and using guides online but I did not succeed as I'm now not able to connect to my Rpi and has no network connection.

thc2018, 2019-01-11:
Install vnstat on RPI3 - it will works fine!

DdraigX, 2019-03-07:
Thanks for the info, I found out that a different software was corrupting my SD Card. Using vnstat now and it's perfect thanks for the info.

closed 2019-03-07 by DdraigX