RPi-Monitor/RPi-Monitor #227: Failed to get readings from DHT 22

tomfruenke, 2018-06-25:
Hi,

I have a little Problem due to I am pretty new to this.

I have set up RPi-Monitor to my Raspberry 3 b+ with Raspbian GNU/Linux 9 (stretch) installed.
To display the DHT 22 temperature sensor I set up a new page on statistics for my serverroom temperature.
The DHT 22 itself works fine, even on the email script I did. 
But RPi-Monitor fails to get the readings from the sensor. I don't know what I did wrong cause I followed your tutorial. Therefore I used the template for DHT sensors.

Here you can see my script:

```
########################################################################
# Extract temperature and humidity from DHT11/DHT22
#  Page: 2
#  Information               Status     Statistics
#  - tempetature             - yes      - yes
#  - humidity                - yes      - yes
#######################################################################
# Add new pages (number 2)
web.status.2.name=Serverroom
web.statistics.2.name=Serverroom

# Get dynamic data
dynamic.1.name=temperature
dynamic.1.source=Adafruit_DHT 22 4
dynamic.1.regexp=t=(\d+)$
dynamic.1.postprocess=$1/1000
dynamic.1.rrd=GAUGE

dynamic.2.name=humidity
dynamic.2.source=Adafruit_DHT 22 4
dynamic.2.regexp=Hum = (\S+)
dynamic.2.postprocess=$1/1000
dynamic.2.rrd=GAUGE

# Add a line of status in page number 2
web.status.2.content.1.name=Temperature web.status.2.content.1.icon=cpu_temp.png
web.status.2.content.1.line.1="Ambient Temperature: <b>"+data.living_room_temp+"&deg;C</b>"
web.status.2.content.1.line.2="Ambient Humidity: <b>"+data.humidity+"%</b>"

# Add a statistic graph (with 2 curves suing separate scales) in page number 2
web.statistics.2.content.1.name=Temperature
web.statistics.2.content.1.graph.1=temperature
web.statistics.2.content.1.graph.2=humidity
web.statistics.2.content.1.ds_graph_options.living_room_temp.label=Temperatur (in °C)
web.statistics.2.content.1.ds_graph_options.humidity.label=Luftfeuchtigkeit (in %)
web.statistics.2.content.1.ds_graph_options.humidity.yaxis=2
#web.statistics.2.content.1.graph_options.legend= { position: "sw"}
web.statistics.2.content.1.graph_options.y2axis={ position: "right", min: 0, max: 100 }

```

In lines 14 and 20 the graph should get values from the DHT22, which is connected to GPIO PIN 4.
But in this step the RPi-Monitor doesn't show any values for Temp und Humidity.

Maybe its only a little problem for what I am blind for.

I hope you can help me with this. I would appreciate it a lot.

cihanoecal, 2018-06-29:
Hi,

Have you tried your regex with `rpimonitord -i` and does it work there?

tomfruenke, 2018-06-29:
My problem was primarily that I can not read correct. -.-' 

In the last paragraph I did not change the variable name, so it would never get to work. 
But I had to change regex and postprocess, so it shows any graph.

I first looked at the output from the script.

It's something like 
`Temp=22.8*C  Humidity=34.3%`

so I had to change my regex and postprocess to
```
dynamic.1.regexp=Temp=(\d+)
dynamic.1.postprocess=$1/1
```
and
```
dynamic.2.regexp=Humidity=(\S+)
dynamic.2.postprocess=$1*1
```

With the post postprocess added like this it will display the numbers in the right format.
So this works fine for me.

@cihanoecal
Is this problem caused by the regex engine ?  
Maybe you have a look at https://regex101.com/ seems a preety good site to test how you could filter your search.

cihanoecal, 2018-06-29:
Hey, good thing that you got yours working :)

Aight lets get to my problem then.
Thx for the tip with regex101.com, but i already knew that site xD 

I was trying to show disk space by mounting point (aka `/media/disk1/`) instead of using the drive partition address (aka `/dev/sda1`). 

So i swapped 
`dynamic.14.regexp=sda1\s+\d+\s+(\d+)`
with
`dynamic.14.regexp=(\d+)\s+\d+\s+\d+%\s+\/media\/disk1`

And it works with the code example above...So yes, I solved my issue... _(Of course you have to swap the regexp for used disk and total disk...)_
My problem was that for some random reason i thought I had to swap all slashes "/" with backslashes "\" for rpi-monitor's regex. **Of course that was wrong**! Now, that i didnt do that step, it worked 🥇 

Pretty sure, RPI-Monitor needs to do better OSI-layer 8 handling xP

Good thing everything works now :) 
This can be closed i guess

tomfruenke, 2018-07-16:
> Pretty sure, RPI-Monitor needs to do better OSI-layer 8 handling xP

I totally agree with you in this point :laughing:

closed 2018-07-16 by tomfruenke