RPi-Monitor/RPi-Monitor #235: Doesn't work on a Pi with Yunohost.

arnauldb, 2018-08-02:
Hello,

I use RPI-monitor on several Pis, but when I installed it on a Pi where Yunohost https://yunohost.org/ is installed, it doesn't work. I can't access http://192.168.1.21:8888/

Any suggestions?

Thank you.

licaon-kter, 2018-08-02:
Error logs?

arnauldb, 2018-08-02:
> Error logs?

This log: /var/log/rpimonitor.log ? : 
```
[2018/08/02-16:43:25] RPi-Monitor has been started
                      Log level can be increased or decreased during runtime
                      by sending signal USR1 or USR2 with the command killall -USRx rpimonitord
```

licaon-kter, 2018-08-02:
And `netstat -tupan | grep LISTEN` shows 8888?

arnauldb, 2018-08-02:
> And netstat -tupan | grep LISTEN shows 8888?

```
des:~# netstat -tupan | grep LISTEN
tcp        0      0 0.0.0.0:993             0.0.0.0:*               LISTEN      1630/dovecot        
tcp        0      0 127.0.0.1:6787          0.0.0.0:*               LISTEN      706/python          
tcp        0      0 127.0.0.1:9091          0.0.0.0:*               LISTEN      707/transmission-da 
tcp        0      0 127.0.0.1:11332         0.0.0.0:*               LISTEN      904/rspamd: main pr 
tcp        0      0 127.0.0.1:11333         0.0.0.0:*               LISTEN      904/rspamd: main pr 
tcp        0      0 0.0.0.0:389             0.0.0.0:*               LISTEN      1146/slapd          
tcp        0      0 127.0.0.1:11334         0.0.0.0:*               LISTEN      904/rspamd: main pr 
tcp        0      0 0.0.0.0:5222            0.0.0.0:*               LISTEN      1501/lua5.1         
tcp        0      0 127.0.0.1:9001          0.0.0.0:*               LISTEN      884/node            
tcp        0      0 127.0.0.1:5290          0.0.0.0:*               LISTEN      1501/lua5.1         
tcp        0      0 0.0.0.0:587             0.0.0.0:*               LISTEN      1542/master         
tcp        0      0 127.0.0.1:6379          0.0.0.0:*               LISTEN      916/redis-server 12 
tcp        0      0 127.0.0.1:5582          0.0.0.0:*               LISTEN      1501/lua5.1         
tcp        0      0 0.0.0.0:143             0.0.0.0:*               LISTEN      1630/dovecot        
tcp        0      0 127.0.0.1:8080          0.0.0.0:*               LISTEN      969/monitorix-httpd 
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      803/nginx: worker p 
tcp        0      0 0.0.0.0:8081            0.0.0.0:*               LISTEN      803/nginx: worker p 
tcp        0      0 0.0.0.0:5269            0.0.0.0:*               LISTEN      1501/lua5.1         
tcp        0      0 0.0.0.0:51413           0.0.0.0:*               LISTEN      707/transmission-da 
tcp        0      0 0.0.0.0:53              0.0.0.0:*               LISTEN      840/dnsmasq         
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      827/sshd            
tcp        0      0 0.0.0.0:8888            0.0.0.0:*               LISTEN      7552/perl           
tcp        0      0 127.0.0.1:5432          0.0.0.0:*               LISTEN      566/postgres        
tcp        0      0 127.0.0.1:61209         0.0.0.0:*               LISTEN      795/python3         
tcp        0      0 0.0.0.0:25              0.0.0.0:*               LISTEN      1542/master         
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      803/nginx: worker p 
tcp        0      0 0.0.0.0:4190            0.0.0.0:*               LISTEN      1630/dovecot        
tcp6       0      0 :::993                  :::*                    LISTEN      1630/dovecot        
tcp6       0      0 ::1:11332               :::*                    LISTEN      904/rspamd: main pr 
tcp6       0      0 ::1:11333               :::*                    LISTEN      904/rspamd: main pr 
tcp6       0      0 :::389                  :::*                    LISTEN      1146/slapd          
tcp6       0      0 ::1:11334               :::*                    LISTEN      904/rspamd: main pr 
tcp6       0      0 :::5222                 :::*                    LISTEN      1501/lua5.1         
tcp6       0      0 :::3306                 :::*                    LISTEN      964/mysqld          
tcp6       0      0 ::1:5290                :::*                    LISTEN      1501/lua5.1         
tcp6       0      0 :::587                  :::*                    LISTEN      1542/master         
tcp6       0      0 ::1:5582                :::*                    LISTEN      1501/lua5.1         
tcp6       0      0 :::143                  :::*                    LISTEN      1630/dovecot        
tcp6       0      0 :::80                   :::*                    LISTEN      803/nginx: worker p 
tcp6       0      0 :::8081                 :::*                    LISTEN      803/nginx: worker p 
tcp6       0      0 :::5269                 :::*                    LISTEN      1501/lua5.1         
tcp6       0      0 :::51413                :::*                    LISTEN      707/transmission-da 
tcp6       0      0 :::53                   :::*                    LISTEN      840/dnsmasq         
tcp6       0      0 :::22                   :::*                    LISTEN      827/sshd            
tcp6       0      0 ::1:5432                :::*                    LISTEN      566/postgres        
tcp6       0      0 :::25                   :::*                    LISTEN      1542/master         
tcp6       0      0 :::443                  :::*                    LISTEN      803/nginx: worker p 
tcp6       0      0 :::4190                 :::*                    LISTEN      1630/dovecot        
tcp6       0      0 :::8000                 :::*                    LISTEN      3643/weechat        
root@abordes:~# 
```

licaon-kter, 2018-08-03:
Firewall, did you open the port?

arnauldb, 2018-08-03:
Thank you licaon-kter . I opened the port 8888 in Yunohost and it is working fine now.

licaon-kter, 2018-08-03:
/close this

closed 2018-08-04 by arnauldb