RPi-Monitor/RPi-Monitor #237: Installation: Keyserver failure

LintHart, 2018-08-17:
Hey Xavier,

I'm pretty new to the RPi world but I'm having trouble with your key server on RPi 3B+ running Raspian Stretch. 

I previously had it running via a manual install, but I couldn't get the CPU temp to show up no matter what I tried (I Also experienced this on a Pi 2b+) . Thought I'd revisit the repo install incase I had better luck.

dirmngr installs fine, then when I attempt the next step:
`sudo apt-key adv --recv-keys --keyserver keyserver.ubuntu.com 2C0D3C0F`

I get the following:
```
Executing: /tmp/apt-key-gpghome.hqUuGFbYgf/gpg.1.sh --recv-keys --keyserver keyserver.ubuntu.com 2C0D3C0F
gpg: keyserver receive failed: Server indicated a failure 
```

Any help would be greatly appreciated thank you

Lint

Leepic, 2019-01-04:
Hello,

Did you try again to pull the key from keyserver.ubuntu.com? I have tested and all is OK:

```
sudo apt-key adv --recv-keys --keyserver keyserver.ubuntu.com 2C0D3C0F
Executing: gpg --ignore-time-conflict --no-options --no-default-keyring --homedir /tmp/tmp.on9ydGft2T --no-auto-check-trustdb --trust-model always --keyring /etc/apt/trusted.gpg --primary-keyring /etc/apt/trusted.gpg --recv-keys --keyserver keyserver.ubuntu.com 2C0D3C0F
gpg: requesting key 2C0D3C0F from hkp server keyserver.ubuntu.com
gpg: key 2C0D3C0F: public key "Xavier Berger <berger.xavier@gmail.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
```

Leepic, 2019-07-20:
@LintHart any update on this issue?

mangss52, 2019-08-24:
Good day Leepic, 
I am having the same issue still and verified your previous post still failed. 

root@raspberrypi:/home/pi# sudo apt-key adv --recv-keys --keyserver keyserver.ubuntu.com 2C0D3C0F
Executing: /tmp/apt-key-gpghome.wbDkHh4dGO/gpg.1.sh --recv-keys --keyserver keyserver.ubuntu.com 2C0D3C0F
gpg: keyserver receive failed: Server indicated a failure

gabrielcbe, 2019-08-29:
+1 on this issue.
i'm having the same problem. but only in a few raspberries (others work fine), and they're supposed to have the same image, since i'm automating the installation with ansible.
anyone know how to deal with this issue yet?

YanWQ-monad, 2019-09-15:
+1

and I manually solved it through:
``` bash
wget -qO - 'https://keyserver.ubuntu.com/pks/lookup?op=get&search=0xe4e362de2c0d3c0f' | sudo apt-key add -
```

XavierBerger, 2026-10-27:
Installation process has been redesigned.
See README.md for details.

closed 2026-10-27 by XavierBerger