RPi-Monitor/RPi-Monitor #238: Problem with InsertHTML

cgaerter, 2018-08-18:
Hi Xavier!

_RPi-Monitor Version: 2.12-r0 running on in total 5 Raspberry Pi with up-to-date Raspbian Lite_ 

Recently I discovered that the top 3 list has disappeared from the main ("Status") page which has previously been shown below the "Scaling governor" line. I tried to investigate and it seems that my configuration is correct:

**/etc/rpimonitor/template/cpu.conf**
(...)
web.status.1.content.1.line.3="Scaling governor: <b>" + data.scaling_governor + "</b>"
web.status.1.content.1.line.4=InsertHTML("/addons/top3/top3.html")

There is a top3.html file present in the /usr/share/rpimonitor/web/addons/top3 folder and its contents are shown correctly via the AddOns -> Top3 menu item.

For testing, I tried to include various html files from various locations with the "InsertHTML" function in the cpu.conf file but in no case the data from the html file is shown on the Status page.

Any help is appreciated.

Thank you and best regards,
Christian

cgaerter, 2018-08-19:
Hi Xavier,

I am sorry, it turned out that this seems to be an issue on my side. :-(

I am using an nginx to forward the requests to the servers (https://dyndomain/1/ to access Pi #1, https://dyndomain/2/ to access Pi #2, and so on). The nginx configuration is as follows:
_location /1/ {
  proxy_pass http://192.168.1.11:8888/;
}_

Now I discovered, that when I access RPi-Monitor via its IP address (192.168.1.11:8888), everything is ok - the top 3 list is shown correctly. Only when I access RPi-Monitor via https://dyndomain/1/, the top 3 list is not shown. Strange ...

I assume this is a configuration issue in ngingx, but I have no clue what I am doing wrong. But I am quite sure, it is not RPi-Monitor's fault. So sorry for blaming you for this!

BTW: If anyone reading this, has an idea what's going wrong here, I am thankful for any hint!

Thank you and best regards,
Christian

cgaerter, 2018-08-20:
Hi Xavier,

it's me again. :-) I have spent the last few hours and finally not only found the reason of the problem but also two possible solutions. I am going to describe what I did - maybe this is useful for someone else.

1) I realized that the top3 addon (and probably others, too) can inject data into the main page using the InsertHTML function. This function however does not take into account that it has been called via a proxy and therefore it tries to access dyndomain/addons/top3/top3.html instead of dyndomain/**1**/addons/top3/top3.html which would be the correct path for Pi #1 in my conficuration (see my posting above).

2) It is obvious that in this way the top3 list cannot be displayed because the call to the webserver returns a 404 error code. I have no idea, why the headers sent by the proxy server are not taken into account by the InsertHTML function, but it appears to be some js/json magic which is beyond my programming skills.

3a) Solution 1: Modify the /etc/rpimonitor/template/cpu.conf file:
web.status.1.content.1.line.4=InsertHTML("/**1**/addons/top3/top3.html")
This has to be done for every RPi-Monitor configuration on each Raspberry Pi with the respective proxy path. The downside is that if you change the proxy path e.g. from dyndomain/1/ to dyndomain/pi1/ you have to modify the configuration file on the correspondent Raspberry Pi.

3b) Solution 2: Add the following lines to the nginx configuration section for each proxy:
_proxy_set_header Accept-Encoding "";
sub_filter_types application/json;
sub_filter '/addons/' '/1/addons/';
sub_filter_once off;_
Line 1 deactivates gzip compression, otherwise nginx cannot perform any text substitutions.
Line 2 indicates that nginx shall do the following substitutions only for this file type.
Line 3 replaces the file path within the file ("/1" is the proxy path for Pi #1)
Line 4 indicates that the replacement has to be done for every occurence.
Using this solution, you do not have to touch the RPi-Monitor configuration files as all modifications are done in one place (in the nginx configuration file on the proxy server).

Best regards,
Christian

Leepic, 2019-01-09:
Can you tell me if the modification made on the issue [#241](https://github.com/XavierBerger/RPi-Monitor/issues/241#issuecomment-452902311) correct your problem?

Leepic, 2019-07-20:
@cgaerter Have you tested the modification of the issue #241?

cgaerter, 2019-07-23:
Yes, changing the link has solved the problems I had. Thank you!

Leepic, 2019-07-27:
@cgaerter perfect! So, you can close this issue :).

closed 2019-07-29 by cgaerter