RPi-Monitor/RPi-Monitor #240: Wrong number of processors

castillo92, 2018-09-15:
In the file:
`/etc/rpimonitor/template/cpu.conf
`

we can see:
```
static.5.name=max_proc
static.5.source=nproc
static.5.regexp=(.*)
**static.5.postprocess=$1 + 1**
```

the proper code should be:
```
static.5.name=max_proc
static.5.source=nproc
static.5.regexp=(.*)
**static.5.postprocess=$1**
```
Because if not, the monitor tell us that the maximun load of CPU is 5 instead of 4

Board: Raspberry Pi 2 B

Leepic, 2019-01-04:
+1. Same for me on a RPi 3B.

```
grep processor /proc/cpuinfo 
processor	: 0
processor	: 1
processor	: 2
processor	: 3
```

RPi-Monitor indicate that is 5 instead of 4:

[image]

Can you submit a pull request?

Leepic, 2019-01-09:
>I just did it.

I can't see your pull request.

>The number of CPU cores are:
>grep processor /proc/cpuinfo processor : 0 processor : 1 processor : 2 processor : 3

>But the output of command "nproc" is 4

Same for me. The two commands are true.

castillo92, 2019-01-10:
> > I just did it.
> 
> I can't see your pull request.
> 
> > The number of CPU cores are:
> > grep processor /proc/cpuinfo processor : 0 processor : 1 processor : 2 processor : 3
> 
> > But the output of command "nproc" is 4
> 
> Same for me. The two commands are true.

Sorry, but I created a fork in my profile. I try to create a Pull Request but I can not edit the file:

https://github.com/XavierBerger/RPi-Monitor/blob/develop/src/etc/rpimonitor/template/cpu.conf

Line 63

Leepic, 2019-01-11:
It's because your branch is not up-to-date with this one. First you need to compare and merge if there are no conflict. After you will be able to edit cpu.conf and submit your pull request :).

moddok, 2019-02-20:
I dont think thats a problem. the 5 is not the count of your cores, but the load of the cpu.
and the load can be bigger than 4.

Leepic, 2019-07-14:
@moddok : it's not a real problem, it's only the display.

You're right, the load can be bigguest than the number of CPU cores but if your load is 5 and you only have 4 cores (physical and logical), the number 5 will be displayed with a red gauge (that indicate a problem).

I don't see a reason of the + 1 in the calcul.

closed 2020-12-22 by castillo92