RPi-Monitor/RPi-Monitor #241: Top3 with reverse-proxy don't work in Status page

shudack, 2018-09-28:
Hi,

Top3 with reverse-proxy don't work in Status page, in localhost:8888 it's work.

Can you help me?

[image]
[image]

Thanks.

Best regards,
Shudack

shudack, 2019-01-04:
I see this on the console.

[image]

Yes, I have access.
[image]

Regards,
Shudack

Leepic, 2019-01-04:
Your console say that `jquery.min.js `can't load. Do you have access with your browser to https://your-domain.tld/rpimonitor/js/jquery.min.js?

The file is at `/usr/share/rpimonitor/web/js/jquery.min.js`.

shudack, 2019-01-07:
Yes, I have access to https://your-domain.tld/rpimonitor/js/jquery.min.js

[image]

Leepic, 2019-01-09:
Hi @shudack 

I have reproduce the problem and find the root cause. I've used the [RPi-Monitor official NginX configuration ](https://xavierberger.github.io/RPi-Monitor-docs/34_autentication.html#authentication-and-secure-access).

In F12 on my web browser I have find that _top3.html_ was not found because the request was made on _https://your-domain.tld/addons/top3/top3.html_ and not _https://your-domain.tld/rpimonitor/addons/top3/top3.html_. So I have just modify cpu.conf:

```diff
-web.status.1.content.1.line.4=InsertHTML("/addons/top3/top3.html")
+web.status.1.content.1.line.4=InsertHTML("./addons/top3/top3.html")
```
The root cause was the dot because without it the request was made on the root domain and not under the directory you are in (/rpimonitor in your case).

Can you tell me it works for you too?

shudack, 2019-01-10:
Hi,

Thanks for help. Works now!

Best regards,
Shudack

closed 2019-01-10 by shudack