RPi-Monitor/RPi-Monitor #242: disabled swap is 100% used

basti12203, 2018-10-25:
Hello,
I have disabled swap on one of my pi's. the rpimonitor shows 100% used.
`cat /proc/meminfo | grep -i swap`
`SwapCached:            0 kB`
`SwapTotal:             0 kB`
`SwapFree:              0 kB`
[image]

Leepic, 2019-01-03:
I have the same issue on my RaspberryPi 3B with RPi-Monitor 2.12-r0 on Raspbian 8 (Jessie).

* When I restart `sudo systemctl restart rpimonitor` with swap disabled:

[image]

* When I enable again the swap with `sudo dphys-swapfile swapon` I can see this:
[image]

* When I restart `sudo systemctl restart rpimonitor`, all is OK:

[image]

As seen above the 100% utilization disappear when I restart rpimonitord. The root cause is because the _swap_total_ value is static and not dynamics in _swap.conf_.

@basti12203 can you replace this lines and tell me if it's OK for you too?

```
dynamic.6.name=swap_total
dynamic.6.source=/proc/meminfo
dynamic.6.regexp=SwapTotal:\s+(\d+)
dynamic.6.postprocess=$1/1024
dynamic.6.rrd=GAUGE
```

I think the value is static for betters performances as RPi-Monitor doesn't need to actualize the _swap_total_ value but only on rpimonitord restart. The _total_swap_ value rarely/never changes on a system. I think it's not a real issue because you just need to restart rpimonitord for the new values take affect.

@XavierBerger What do you think?

basti12203, 2019-07-10:
yes that works. thanks a lot. i have also change /usr/share/rpimonitor/web/js/rpimonitor.utils.js
` function Percent(value,total){
-  return (100*value/total).toFixed(2)+"%";
+  if (total > 0) {
+       return (100*value/total).toFixed(2)+"%";
+  } else { return (0)+"%"; }
 }`
to fix NaN% Output

closed 2019-07-10 by basti12203

TupaNegreiros, 2021-02-03:
I'm using `dphys-swapfile`, changing static to dynamic worked here.
It's needed to also change `dynamic.8.postprocess=$this->{'static'}->{'swap_total'} - ($1/1024)` to `dynamic.8.postprocess=$this->{'dynamic'}->{'swap_total'} - ($1/1024)`