RPi-Monitor/RPi-Monitor #250: Temperature not being measured

Tijnert, 2019-01-10:
Hi,

Installed RPI-Monitor with the default temperature.conf file.
The Temperature doesn't show up .

content conf file:

```
########################################################################
# Extract CPU Temperature information
#  Page: 1
#  Information               Status     Statistics
#  - cpu temperature         - yes      - yes
#
# Note: Rounding temperatures to two digits is done with the
#  post-processiing formula: sprintf("%.2f", $1/1000)
#  It is possible to roud to 1 digit with sprintf("%.1f", $1/1000)
########################################################################
dynamic.1.name=soc_temp
dynamic.1.source=/sys/devices/virtual/thermal/thermal_zone0/temp
dynamic.1.regexp=(.*)
dynamic.1.postprocess=sprintf("%.2f", $1/1000)
dynamic.1.rrd=GAUGE

web.status.1.content.1.title="Temperature"
web.status.1.content.1.icon=cpu_temp.png
web.status.1.content.1.line.1="CPU Temperature: <b>"+data.soc_temp+"&deg;C</b>"
web.status.1.content.1.line.1=JustGageBar("CPU Temperature", data.soc_temp+"°C"$
web.status.1.content.1.line.1=JustGageBar("Temperature", "°C", 40, data.soc_tem$

web.statistics.1.content.1.title="Temperature"
web.statistics.1.content.1.graph.1=soc_temp
web.statistics.1.content.1.ds_graph_options.soc_temp.label=Core temperature (°C)
```

[image]

wuffer, 2019-01-11:
I just installed it, and i got the same problem

thc2018, 2019-01-11:
This problem was solved here - https://github.com/XavierBerger/RPi-Monitor/issues/245

licaon-kter, 2019-01-12:
I had to edit **/etc/rpimonitor/template/temperature.conf**:
```
#dynamic.1.postprocess=sprintf("%.2f", $1/1000)
dynamic.1.postprocess=$1/1000
```

wuffer, 2019-01-12:
> #dynamic.1.postprocess=sprintf("%.2f", $1/1000) dynamic.1.postprocess=$1/1000

Thanks. That seem to do the trick :)

reubadoob, 2019-06-09:
@licaon-kter Thanks! Worked perfectly.

closed 2020-06-01 by XavierBerger

dineiar, 2021-04-15:
Using `dynamic.1.postprocess=$1/1000` according to @licaon-kter's [suggestion](https://github.com/XavierBerger/RPi-Monitor/issues/250#issuecomment-453717615) causes the temperature to display three-digit numbers and sometimes overflow the gauge area (see [this example](https://github.com/XavierBerger/RPi-Monitor/issues/303#issuecomment-788188492)).

To fix this, I changed the web to show only two digits. I.e. change from:
```
web.status.1.content.1.line.1=JustGageBar("CPU Temperature", data.soc_temp+"°C", 40, data.soc_temp, 80, 100, 80)
```
to:
```
web.status.1.content.1.line.1=JustGageBar("CPU Temperature", data.soc_temp+"°C", 40, data.soc_temp.toFixed(2), 80, 100, 80)
```
I added `.toFixed(2)` in the number displayed in the center of the gauge.

racedowling, 2022-03-16:
@dineiar 
I added .toFixed(2) as above but it returns an error. 

JustGageBar("Voltage","V",0,data.volts.toFixed(2),4,100,80,"",50,75) -> TypeError: data.volts.toFixed is not a function

dineiar, 2022-03-16:
@racedowling, how is the `volts` information defined in your config file?  
The [original CPU template](https://github.com/XavierBerger/RPi-Monitor/blob/develop/src/etc/rpimonitor/template/cpu.conf) only contains a field called `cpu_voltage`.

racedowling, 2022-03-16:
@dineiar
I used volts so that I wouldn't have to change the working cpu_voltage.  I went through and changed everything to cpu_voltage but the result was the same. 

" + JustGageBar("Voltage","V",0,data.cpu_voltage.toFixed(2),4,100,80,"",50,75) -> TypeError: data.cpu_voltage.toFixed is not a function

I don't know if this makes a difference but this is how I'm setting the variable:
dynamic.3.name=cpu_voltage
dynamic.3.source=vcgencmd measure_volts core
dynamic.3.regexp=volt\=(.*)V
dynamic.3.postprocess=$1
dynamic.3.rrd=GAUGE

The other fix - int($1/100+0.5)/100 works in rpimonitord -i but in the config file it only returns 0.

dineiar, 2022-03-16:
It's ok to use your own variables, I also do this all the time.  
My guess is that the error is because your variable is being passed to JavaScript as an `string` and `toFixed` is a method of the `number` prototype. I see two options to fix it:
1. Force storing the number as a number. I think you can do this by changing your `postprocess` to perform any computation (e.g. `postprocess=$1/1` should be enough).
2. Cast the string to number before calling `toFixed`, i.e. `parseFloat(data.cpu_voltage).toFixed(2)`

IMHO, I prefer the first option, but maybe you will need to erase your database if you go with it, because I'm not sure if the backend stores schema/column types.

racedowling, 2022-03-16:
@dineiar 
The divide by 1 worked! (Tip of my hat to you).  Thanks, this has been driving me crazy since upgrading from 12.
I earlier tried *1 which did not. Strange error message for a type mismatch.