RPi-Monitor/RPi-Monitor #252: Add CPU Frequency Statistics

Trotter73, 2019-01-17:
Hi,

How easy would it be to add the CPU Frequency to the statistics page, can I use the interactive helper for this ? If not where would I start ?

Thanks in advance.

moddok, 2019-02-20:
Hey,

its very easy. just edit the cpu.conf file and add the following lines in the statistics block:
`web.statistics.1.content.1.graph.X=cpu_frequency
web.statistics.1.content.1.ds_graph_options.cpu_frequency.label=CPU Frequency (MHz)
web.statistics.1.content.1.graph_options.y2axis={ position: "right" }`

Change the X in the code to the right number of your graphs.
the last line makes the units stand on the right side

Trotter73, 2019-02-21:
Huge thanks for this, very much appreciated ! I will try tonight.

moddok, 2019-02-21:
I forgot one thing. You have to add the rrd type to the dynamic block.
So it have to look something like this:
`dynamic.1.rrd=GAUGE
`
```
dynamic.1.name=cpu_frequency
dynamic.1.source=/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq
dynamic.1.regexp=(.*)
dynamic.1.postprocess=$1/1000
dynamic.1.rrd=GAUGE
```

Trotter73, 2019-02-21:
Hi,

Thanks for the followup, getting the idea now, I also had to add this to get the units showing on the right... All working, perfect, thanks again !

web.statistics.1.content.1.ds_graph_options.cpu_frequency.yaxis=2

closed 2019-02-22 by Trotter73