RPi-Monitor/RPi-Monitor #255: DS18B20 temp sensor at RPi-Monitor

lukaszertel, 2019-02-21:
Dear RPi-Monitor Developers,
Could you paste temperature config regarding DS18B20 temp sensor please?
I found tutorials but only for DHT sensors.
I'm using RPi-Monitor Version: 2.12-r0

Cheers!
Lucas

Leepic, 2019-07-14:
Hello,

I've found some explanations on the [RPi-Monitor docs](https://xavierberger.github.io/RPi-Monitor-docs/32_sensors.html#ds18b20-1wire-temperature-sensor).

Here the complete setup on the [Adafruit website lesson 11](https://learn.adafruit.com/adafruits-raspberry-pi-lesson-11-ds18b20-temperature-sensing?view=all)

Tell me if it works. If so, give us your configuration and I will submit a pull request for updating the docs (or you can do it if you want).

Leepic, 2019-07-27:
@lukaszertel HAve you try the links in my previous comment? If it works, you can close this issue :).

closed 2020-06-01 by XavierBerger