RPi-Monitor/RPi-Monitor #257: Arch package update list error

Tigrejor, 2019-02-25:
When trying to run `sudo /etc/init.d/rpimonitor update` on arch it gives me the error /lib/lsb/init-functions doesn't exist.

> /etc/init.d/rpimonitor: línea 14: /lib/lsb/init-functions: No existe el fichero o el directorio

I know Arch is an unsupported OS, but could you give any support on this?
Thanks in advance.

heitorPB, 2019-04-10:
Same problem here using aarch64

Leepic, 2019-07-28:
Hello,

This error appears because ArchLinux is not [LSB](https://en.wikipedia.org/wiki/Linux_Standard_Base) compliant, so _/lib/lsb/init-functions_ is not present. Using a systemd unit will works.

`sudo /etc/init.d/rpimonitor update` will fail because it's based on the apt-get system manager or checkupdates [See the file /usr/share/rpimonitor/scripts/updatePackagesStatus.pl](https://github.com/XavierBerger/RPi-Monitor/blob/master/src/usr/share/rpimonitor/scripts/updatePackagesStatus.pl). When this perl script is run it produce this error:
_Error: neither checkupdates nor aptitude seem to be available_

Currently, some others features may not works on ArchLinux.

Tigrejor, 2019-07-28:
Hi.
So from what you said, I should change it so that it uses systemd?
Also, if I were to try and make it work, I should modify that perl script so it uses pacman or wouldn't it work since it is pretty different?
Thanks.

heitorPB, 2019-07-28:
IMO, it should use something generic to check for updates, instead of a specific tool like `apt`. Maybe check if there's a new git tag or something like that.

Tigrejor, 2019-07-28:
I don't think that would work since I think there could be an update ready in apt but the same update not being yet on pacman, correct me if I'm wrong anyways.

Leepic, 2019-08-16:
@heitorPB I don't known if there is something generic but It's impossible to based the check on git tag because the packages are not git based.

@Tigrejor I review the script and made somes tests and a systemd service is already present on Arch (See /usr/lib/systemd/system/rpimonitord.service). I have remove the _PIDFile_ line, otherwise on start the operation will timeout
`Job for rpimonitord.service failed because a timeout was exceeded.
See "systemctl status rpimonitord.service" and "journalctl -xe" for details.
`
Don't forget to run `systemctl daemon-reload` after the modification.

rpimonitord start sucessfully with the message in log `rpimonitord.service: Failed to parse PID from file /run/rpimonitord.pid: Invalid argument`.

Regarding the update, there is a bash script named _checkupdates_ in the package _pacman-contrib_  and RPi-Monitor already works with it (See [the line 23 of /usr/share/rpimonitor/scripts/updatePackagesStatus.pl](https://github.com/XavierBerger/RPi-Monitor/blob/master/src/usr/share/rpimonitor/scripts/updatePackagesStatus.pl#L23)) ;-).

heitorPB, 2019-08-17:
> @heitorPB I don't known if there is something generic but It's impossible to based the check on git tag because the packages are not git based.

I never got the reason for a package to check if it is up to date. The latest version might not be stable... Why not drop this feature?

XavierBerger, 2020-06-01:
**RPI-Monitor** is a tool you can configure up to your need.
I'll let you customize it to your need and the close this issue.

closed 2020-06-01 by XavierBerger