RPi-Monitor/RPi-Monitor #259: Monitoring Networkspeed with Iperf

tmsd2001, 2019-03-28:
I try to Monitor Network speed with RPI-Monitor and Iperf, but I get this error:
Warning: unable to close filehandle FEED properly.

RPI-Monitor Release is 2.12-r0

This is my iperf.conf file:

`dynamic.17.name=Iperf
dynamic.17.source=iperf -p 4711 -c 192.168.0.254 | grep -Po '(?<=,)[0-9]+$|[0-9.]*(?= Mbits\/sec)|(?<=local\/sender )[^%]*' | paste -d, -
dynamic.17.regexp=(.*)
dynamic.17.postprocess=$1 * 1
dynamic.17.rrd=GAUGE

web.status.1.content.17.name=Iperf
web.status.1.content.17.icon=wifi.png
web.status.1.content.17.line.1='Datenrate: <b>' + KMG(data.Iperf,'M')

web.statistics.1.content.17.name=Iperf
web.statistics.1.content.17.graph.1=Iperf
web.statistics.1.content.17.ds_graph_options.Iperf.label=Mbyte/sec
web.statistics.1.content.17.ds_graph_options.Iperf.lines={ fill: true }
web.statistics.1.content.17.ds_graph_options.Iperf.color="#FF7777"
`

tmsd2001, 2019-05-19:
my hotfix is I put the command in a bash file and write the output to a file.
In RPiMonitor I use as Source dynamic.17.source=tail -n1 iperf_file.txt.
This is working.

Leepic, 2019-07-20:
I think it's because the iperf command take more than 5 seconds to execute (the default RPi-Monitor).

If you don't want to use an intermediate file, you need to modify RPi-Monitor daemon like this:

* /etc/rpimonitor/daemon.conf

`daemon.timeout=30`

Then restart rpimonitor.

Leepic, 2019-07-27:
@tmsd2001 I think that this issue is resolved. So, can you close it, please? thanks!

closed 2019-07-27 by tmsd2001