RPi-Monitor/RPi-Monitor #260: Packaged Version has not SNMP

pssc, 2019-03-28:
Installed per instructions missing snmp functionality, in packaged version.

XavierBerger, 2020-06-01:
Bug description is too short to understand what could happen

ewallquist, 2020-12-16:
Could be the same thing I just noticed. When installing following the instructions there is no /etc/snmp/snmpd.conf.rpimonitor  installed, neither is /usr/bin/rpimonitord-snmp . So when trying to access snmp-data all I get is 

> SNMPv2-SMI::enterprises.54321.42 = No more variables left in this MIB View (It is past the end of the MIB tree)

ewallquist, 2020-12-16:
Also, in the source it says
>   -l, --logfile     Logfile (ex: /var/log/rpimonitor.log)
  -m, --mib         Get MIB for current configuration
  -n, --noserver    Don't start embeded web server

but on the installed version
>   -l, --logfile     Logfile directory and prefix (ex: /var/log/rpimonitor)
  -n, --noserver    Don't start embeded server
so there is no flag to extract the mib

and, finally, after some more checking, the installed version from the repo is 

> 

> /usr/bin/rpimonitord version 2.12-r0