RPi-Monitor/RPi-Monitor #265: Version not updating

mushu999, 2019-05-12:
Web interface Help/About:  Version: 2.12-r0 by Xavier Berger 
```
>sudo apt-cache policy rpimonitor
rpimonitor:
  Installed: 2.12-r0
  Candidate: 2.12-r0
  Version table:
 *** 2.12-r0 500
        500 http://giteduberger.fr rpimonitor/ Packages
        100 /var/lib/dpkg/status

> sudo cat /etc/apt/sources.list.d/rpimonitor.list
# RPi-Monitor official repository
deb http://giteduberger.fr rpimonitor/

>sudo apt update && sudo apt upgrade -y
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Ign:3 http://giteduberger.fr rpimonitor/ InRelease
Get:4 http://giteduberger.fr rpimonitor/ Release [1,396 B]
Fetched 1,396 B in 2s (676 B/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
Is not updating to r1 as shown available in your Release Notes and I don't want to uninstall and reinstall and lose my code modifications. Yes, I followed the Installation instructions to update my repositories and yes I ran the monitor with "update" parameter afterwards. What is the problem?

Leepic, 2019-07-08:
The version 2.12-r1was not released yet (flag [UNRELEASED](https://github.com/XavierBerger/RPi-Monitor-deb/blob/develop/debian/changelog#L16)).

In [the official repository](http://giteduberger.fr/rpimonitor/Packages) it's mentionned the version 2.12-r0, that's the latest realase in the master branch. Like display in your ```apt-cache policy rpimonitor```  command.

Leepic, 2019-07-27:
@mushu999 Have you read my previous comment? It answer your question or you have another question? If not, you can close this case.

mushu999, 2019-08-04:
So the release notes were incorrect? Okay. Perhaps they should be fixed to avoid confusion in the future.

closed 2019-08-04 by mushu999

Leepic, 2019-08-16:
I'm agree with you, there are somes problems in the changelog file. [Line 1](https://github.com/XavierBerger/RPi-Monitor-deb/blob/develop/debian/changelog#L1), there is no indication about the concerned release (maybe 2.13-beta6, the latest).

2.11-r5 is mark UNRELEASE but the [deb package](https://github.com/XavierBerger/RPi-Monitor-deb/tree/develop/packages) is available, same for 2.10-1 but nothing in changelogs about 2.11-r4…