RPi-Monitor/RPi-Monitor #266: Network monitoring

Therealjosephchrzempiec, 2019-05-20:
Hello I'm still kinda new to the whole monitoring and programming. But i was having a hard to time to enable network monitoring can you please help me out? Second question is it possible to something like shellinabox to rpi-monitoring?? 

Joseph

Therealjosephchrzempiec, 2019-05-20:
Hello i did figure out the whole shellinabox it looks like you guys build one in. Only problem I see what that is when i switch to a top menu button it ask if i want to leave shellinabox and logs me out. Is there a way to stop that. So when i go back into that shellinabox tab it can remains without having to log back in all the time?

Leepic, 2019-07-08:
Hello,

What is your need for network monitoring?

Regarding ShellInABox I've the same problem and the only workaround I've find is to open a second tab with this addon only.

jwtaylor310, 2019-08-13:
The distribution version of RPi-Monitor is set up to display the number of bytes sent and received on a network interface.  To enable that, you need to edit the network.conf file.  To do so, open a Terminal screen and enter the following command:

sudo nano /etc/rpimonitor/template/network.conf

That will open up the file in the "nano" text editor.  Use the arrow keys to move down and delete the "#" character at the beginning of each line starting with the one that says: 

dynamic.10.name=net_received  

This will enable monitoring for the "eth0" device which in most installations will be the wired network interface.  If you instead want to monitor the WiFi interface, look through the configuration file and replace each appearance of "eth0" with the name of your WiFi interface, usually "wlan0".

When you are finished editing the file, press CTRL-X followed by "y" to save the changes.  Restart RP-Monitor (or reboot the Pi) and the networking monitor should be active.

Good luck!

closed 2020-06-01 by XavierBerger