RPi-Monitor/RPi-Monitor #267: E: The repository 'http://giteduberger.fr rpimonitor/ Release' does no longer have a Release file.

Firesphere, 2019-05-20:
Domain seems down

closed 2019-06-20 by XavierBerger

thijstriemstra, 2019-08-05:
same issue just now: 
```
Genegeerd:3 http://giteduberger.fr rpimonitor/ InRelease                                
Fout:4 http://giteduberger.fr rpimonitor/ Release               
  503  Service Unavailable
```

Firesphere, 2019-08-16:
Why is this closed without any response?

guillaumedavidphd, 2019-11-20:
I'd love to have the answer to this question.

guillaumedavidphd, 2019-11-20:
Okay, I edited `/etc/apt/sources.list.d/rpimonitor.list` and changed the following line from

`deb http://giteduberger.fr rpimonitor/`

to

`deb https://www.giteduberger.fr rpimonitor/`

That was enough to solve the problem and install `rpimonitor`.

axl12ap, 2020-03-24:
I'm trying to install RPI-monitor using the method on webpage but same error now appears on my raspberry.

when I get the file it contains "http" nor "https", so I tried to edited in https but nothing change, same error of thread title.

Help me please

srg-z, 2020-06-14:
custom installation via DEB-pack still working, step-by-step guide correct, just installed.

https://xavierberger.github.io/RPi-Monitor-docs/12_custom_installation.html

icolwell, 2020-12-06:
Thanks @guillaumedavidphd!
If anyone is looking for an updated/working one-line install script, you can try this:
```
curl -sSL https://raw.githubusercontent.com/icolwell/install_scripts/master/rpimonitor_install.bash | bash
```

Source:
https://github.com/icolwell/install_scripts/blob/master/rpimonitor_install.bash

ardinusawan, 2022-10-17:
This did the trick https://forums.raspberrypi.com/viewtopic.php?t=255932#p1823865

```
# RPi-Monitor official repository
deb https://www.giteduberger.fr rpimonitor/
```