RPi-Monitor/RPi-Monitor #268: External links going to external tab in browser?

Therealjosephchrzempiec, 2019-05-20:
Hello i was wondering if it possible to have a external link in the top bar going to a external tab in the browser. In html it would be <a href="https://www.google.com" target="_blank">google</a> But in raspberry pi and how this is setup I have no clue if this is possible. Can someone help me to figure this out and help to put this in?

Joseph

Leepic, 2019-07-08:
Hello,

In my knowledge, the only solution is using [Friends](https://xavierberger.github.io/RPi-Monitor-docs/27_configuration_templates.html#friends) link, like this :

- /etc/rpimonitor/data.conf

```
web.friends.1.name=GitHub
web.friends.1.link=https://github.com/XavierBerger/RPi-Monitor/issues/268
```
It's not exactly what you're looking for but it works.

Leepic, 2019-07-27:
@Therealjosephchrzempiec have you try the Friends solution?

XavierBerger, 2020-06-01:
No anwser ... closing

closed 2020-06-01 by XavierBerger