RPi-Monitor/RPi-Monitor #269: InsertHTML cross-over

patricktokeeffe, 2019-06-27:
I have a status page configured with two `InsertHTML` directives:
```
web.status.2.content.4.name=Alarms
web.status.2.content.4.icon=warning.png
web.status.2.content.4.line.1="<pre>"+InsertHTML("/data/ebam/latest/Alarm_Report")+"</pre>"

web.status.2.content.5.name=Settings
web.status.2.content.5.icon=usb_hdd.png
web.status.2.content.5.line.1="<pre>"+InsertHTML("/data/ebam/latest/Settings_Report")+"</pre>"
```

A small typo (`Alarms_Report` instead of `Alarm_Report`) resulted in one of my HTML sources being unavailable:

[image]

Instead of ignoring the missing HTML file, it is silently replaced with the parameter provided to my *other* `InsertHTML` directive (i.e. the result is duplicate insertions):

[image]

A hard page refresh will produce the expected result (blank entry for Alarms) but the "Auto refresh page status" feature will repopulate the alarm field with the wrong file.

Leepic, 2019-07-14:
Have you try to restart rpimonitor and open it's webUI in a private tab? Maybe it can be caused by the cache...

htcfreek, 2020-03-23:
Hello.

I have encountered the same problem with the following pages:
- https://api4.ipify.org
- https://api6.ipify.org

My config:
`[...]line.1="Public IPv4: "+InsertHTML("https://api4.ipify.org")
[...]line.2="Public IPv6: "+InsertHTML("https://api6.ipify.org")`

Right after setting config, reastarting the deamon and loading rpimonitor-page in browser, I get shown the same value for both addresses.