RPi-Monitor/RPi-Monitor #273: Raspberry Pi 4 CPU identification incorrect

nthijssen, 2019-07-09:
Rpi monitor signals on a Raspberry Pi 4 as CPU: BCM2835

Expected: BCM2711B0
Rpimonitor version used: 2.12R0

Leepic, 2019-07-22:
You're right as the [official documentation](https://www.raspberrypi.org/documentation/hardware/raspberrypi/) say but as indicate on [raspberrypi.org](https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md), starting from kernel 4.9, all Hardware indicate BMC2835:

> Note: As of the 4.9 kernel, all Pis report BCM2835, even those with BCM2836 and BCM2837 processors. You should not use this string to detect the processor.

The revision codes only show the model of RaspberryPi, not directly the processor used. And the model name can be found in _/proc/device-tree/model_.

The processor can be found in the file _/proc/device-tree/compatible_ but I don't know if it's the right way and if this file will be stable in the future.

Here the output from:
- RaspberryPi Model B+ revv 1.2:

```shell
grep -Eao '(bcm)[0-9A-Z]+' /proc/device-tree/compatible
bcm2835
```

- RaspberryPi Model 3 B Rev 1.2:

```shell
grep -Eao '(bcm)[0-9A-Z]+' /proc/device-tree/compatible
bcm2837
```

RSlegers, 2020-04-17:
I have changed my cpu.conf into the following:
`########################################################################
# Extract CPU information
#  Page: 1
#  Information               Status     Statistics
#  - cpu frequency           - yes      - no
#  - cpu voltage             - yes      - no
#  - cpu load 1, 5, 15       - yes      - yes
#  - cpu scaling governor    - yes      - no
########################################################################
dynamic.1.name=cpu_frequency
dynamic.1.source=/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq
dynamic.1.regexp=(.*)
dynamic.1.postprocess=$1/1000
dynamic.1.rrd=GAUGE

dynamic.2.name=cpu_voltage
dynamic.2.source=vcgencmd measure_volts core
dynamic.2.regexp=(\d+.\d+)V
dynamic.2.postprocess=
dynamic.2.rrd=

dynamic.3.name=load1,load5,load15
dynamic.3.source=/proc/loadavg
dynamic.3.regexp=^(\S+)\s(\S+)\s(\S+)
dynamic.3.postprocess=
dynamic.3.rrd=GAUGE

dynamic.4.name=scaling_governor
dynamic.4.source=/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
dynamic.4.regexp=(.*)
dynamic.4.postprocess=
dynamic.4.rrd=

static.9.name=max_proc
static.9.source=nproc
static.9.regexp=(.*)
static.9.postprocess=$1 + 1

static.10.name=max_cpu_frequency
static.10.source=/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq
static.10.regexp=(.*)
static.10.postprocess=$1/1000

static.11.name=min_cpu_frequency
static.11.source=/sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_min_freq
static.11.regexp=(.*)
static.11.postprocess=$1/1000

web.status.1.content.1.name=CPU
web.status.1.content.1.icon=cpu.png
#web.status.1.content.1.line.1="Loads: <b>" + data.load1 + "</b> [1min] - <b>" + data.load5 + "</b> [5min] - <b>" + data.load15 + "</b> [15min]"
web.status.1.content.1.line.1=JustGageBar("Load", "1min", 0, data.load1, data.max_proc, 100, 80)+" "+JustGageBar("Load", "5min", 0, data.load5, data.max_proc, 100, 80)+" "+JustGageBar("Load", "15min", 0, data.load15, data.max_proc, 100, 80)
web.status.1.content.1.line.2=JustGageBar("CPU frequency", "MHz", 0, data.cpu_frequency, data.max_cpu_frequency)
web.status.1.content.1.line.3="CPU frequency: <b>" + data.cpu_frequency + " MHz</b> Voltage: <b>" + data.cpu_voltage + " V</b>"
web.status.1.content.1.line.4="Scaling governor: <b>" + data.scaling_governor + "</b>"
#web.status.1.content.1.line.5=InsertHTML("/addons/top3/top3.html")

web.statistics.1.content.1.name=CPU Loads
web.statistics.1.content.1.graph.1=load1
web.statistics.1.content.1.graph.2=load5
web.statistics.1.content.1.graph.3=load15
web.statistics.1.content.1.ds_graph_options.load1.label=Load 1min
web.statistics.1.content.1.ds_graph_options.load5.label=Load 5min
web.statistics.1.content.1.ds_graph_options.load15.label=Load 15min

web.statistics.1.content.11.name=CPU Frequency
web.statistics.1.content.11.graph.1=cpu_frequency
web.statistics.1.content.11.graph.2=max_cpu_frequency
web.statistics.1.content.11.graph.3=min_cpu_frequency
web.statistics.1.content.11.ds_graph_options.cpu_frequency.label=Current CPU Frequency
web.statistics.1.content.11.ds_graph_options.max_cpu_frequency.label=Maximum CPU Frequency
web.statistics.1.content.11.ds_graph_options.min_cpu_frequency.label=Minimum CPU Frequency
`
It shows the correct CPU and adds statistics for the CPU Frequency (Mostly for the RPI4)

Hope you find it usefull

Leepic, 2020-04-19:
@RSlegers and in the Version section, what's your Processor hardxare information with RPi 4 (BCM)?

RSlegers, 2020-04-19:
```
Model: Raspberry Pi 4 Model B Rev 1.1 (b03111)
Processor: BCM2711 Memory : 2048Mb
Distribution: Raspbian GNU/Linux 10 (buster)
Kernel version: Linux 4.19.97-v7l+ armv7l (#1294)
Package(s): 0 upgradable(s) 
```
This is my output, i only forgot to mention you also need to install python-gpiozero or python3-gpiozero

Ehhh, i posted the wrong conf file, i see now, the one i posted is cpu.conf, but it must be version.conf, i will post here:
```
########################################################################
# Extract System information
#  Page: 1
#  Information               Status     Statistics
#  - distribution name       - yes      - no
#  - kernel version          - yes      - no
#  - firmware version        - yes      - no
#  - processor model         - yes      - no
#  - num of pkg upgradable   - yes      - no
#  - list of pkg upgradable  - yes      - no
########################################################################
static.1.name=distribution
static.1.source=/etc/os-release
static.1.regexp=PRETTY_NAME.\"(.*)\"
static.1.postprocess=

static.2.name=kernel_version
static.2.source=uname -msr
static.2.regexp=(.*)
static.2.postprocess=

static.3.name=firmware
static.3.source=/proc/version
static.3.regexp=(#\d+)
static.3.postprocess=

static.4.name=processor
static.4.source=pinout -m
static.4.regexp=(?:SoC)\s+: (.*)
static.4.postprocess=

static.5.name=revision
static.5.source=/proc/cpuinfo
static.5.regexp=(?:Revision)\s+: (.*)
static.5.postprocess=

static.6.name=model
static.6.source=/proc/cpuinfo
static.6.regexp=(?:Model)\s+: (.*)
static.6.postprocess=

static.7.name=memory
static.7.source=pinout -m
static.7.regexp=(?:RAM)\s+: (.*)
static.7.postprocess=

dynamic.1.name=upgrade
dynamic.1.source=/var/lib/rpimonitor/updatestatus.txt
dynamic.1.regexp=(\d+ upgradable\(s\)|.*&nbsp;.*)
dynamic.1.postprocess=
dynamic.1.rrd=

dynamic.2.name=packages
dynamic.2.source=/var/lib/rpimonitor/updatestatus.txt
dynamic.2.regexp=^\s+(.*)\s+\d+ upgradable
dynamic.2.postprocess=
dynamic.2.rrd=

web.status.1.content.1.name=Version
web.status.1.content.1.icon=version.png
web.status.1.content.1.line.1='Model: <b>' + data.model + ' (<i>' + data.revision + ')</i></b>'
web.status.1.content.1.line.2='Processor: <b>' + data.processor + '</b>   Memory : <b>' + data.memory + '</b>'
web.status.1.content.1.line.3='Distribution: <b>'+ data.distribution + '</b>'
web.status.1.content.1.line.4='Kernel version: <b>' + data.kernel_version + ' (<i>' + data.firmware + ')</i></b>'
web.status.1.content.1.line.5='Package(s): <b>' + data.upgrade + '</b>' + ShowInfo('packages','Package(s) upgradable(s)',data.packages)

```

closed 2020-06-01 by XavierBerger