RPi-Monitor/RPi-Monitor #278: Manage Internet Broadband with speedtest-cli

IssueFindings, 2019-07-19:
Hello,
I try to modify network.conf to track internet broadband with the python command _speedtest-cli_. That doesn't work, I make a mistake but I don't find where. So, could you help me ? Have a nice day.

Please find below my code.
`dynamic.10.name=net_received
dynamic.10.source=speedtest-cli --simple | grep Download | cut -d" " -f2
dynamic.10.regexp=(.*)
dynamic.10.rrd=GAUGE
dynamic.10.interval=6
etc.`

Leepic, 2019-07-20:
Hello,

The problem is that the command take too much time to execute (about 30 secondes), so RPi-Monitor can't get the value and displays NaN as a results.
You need to increase the default (5 secondes) timeout for KPI extraction:

* /etc/rpimonitor/daemon.conf

`daemon.timeout=30`

You also need to tell to RPi-Monitor that value is in MB:

* Before

```
web.status.1.content.8.line.1="Ethernet Sent: <b>"+KMG(data.net_send)+"<i class='icon-arrow-up'></i></b> Received: <b>"+KMG(Math.abs(data.net_received)) + "<i class='icon-arrow-down'></i></b>"
```

* After

```
web.status.1.content.8.line.1="Ethernet Sent: <b>"+KMG(data.net_send)+"<i class='icon-arrow-up'></i></b> Received: <b>" +KMG(data.net_received, 'M')+"<i class='icon-arrow-down'></i></b>"
```

* Then, restart RPi-Monitor

`systemctl restart rpimonitor`

IssueFindings, 2019-07-20:
Hello,

I update daemon.conf and my network.conf. To help other, please find below the entire code of network.conf (I try to simplify it :-)). Just a last question regarding graph options. I don't know why the graph displayed y axis between -1 and 1. Why ? How I can change it ?

`dynamic.10.name=net_received
dynamic.10.source=speedtest-cli --simple | grep Download | cut -d" " -f2
dynamic.10.regexp=(.*)
dynamic.10.rrd=GAUGE
dynamic.10.interval=6

dynamic.11.name=net_send
dynamic.11.source=speedtest-cli --simple | grep Upload | cut -d" " -f2
dynamic.11.regexp=(.*)
dynamic.11.rrd=GAUGE
dynamic.11.interval=6

web.status.1.content.8.name=Internet
web.status.1.content.8.icon=network.png
web.status.1.content.8.line.1="Public IP: <b>" + InsertHTML("http://ifconfig.me/ip") + "</b>"
web.status.1.content.8.line.2='Upload: <b>' + KMG(data.net_send, 'M') + '</b> Download: <b>' + KMG(data.net_received, 'M') + '</b>'

web.statistics.1.content.2.name=Internet
web.statistics.1.content.2.graph.1=net_send
web.statistics.1.content.2.graph.2=net_received
web.statistics.1.content.2.ds_graph_options.net_send.label=Upload (MB)
web.statistics.1.content.2.ds_graph_options.net_send.color="#FF7777"
web.statistics.1.content.2.ds_graph_options.net_received.label=Download (MB)
web.statistics.1.content.2.7ds_graph_options.net_received.color="#77FF77"
`
Have a nice day.

Leepic, 2019-07-20:
Hello,

I can't reproduce the issue with your configuration… I don't have any value on the graphs but on the status page it works. Can you give me a screenshot of your problem?

Thanks.

IssueFindings, 2019-07-21:
Hello,
you are right, it's better to put a screenshot. Please see below.

[image]

As you can see, "y" axis is limited between -1 and 1. So it can display results. What do you think ?

Have a nice day.

Leepic, 2019-07-27:
@IssueFindings I've the same problem. I think that it's related to the units (Mbit/s vs bytes) because with the default source file (in bytes) it works well.

Currently, I've no solution. I'll keep you informed ;).

IssueFindings, 2019-07-28:
Many thanks for your help and your time.

Leepic, 2020-01-01:
Hello @IssueFindings,

I'm back with a working solution (status and statistics pages). Only the units are false on the status page (kB or MB instead of Kb or Mb).

Here my working configuration:

Extract the KPI information every 60 secondes (**interval=6**). It's useless every minutes, use **interval=360** instead for every hour or **interval=8640** for every day only.

- /etc/rpimonitor/template/network.conf

```
dynamic.10.name=net_received
dynamic.10.source=speedtest-cli --simple --no-upload | grep Download | cut -d" " -f2
dynamic.10.regexp=(.*)
dynamic.10.postprocess=$1*131*8
dynamic.10.rrd=GAUGE
dynamic.10.interval=6
dynamic.10.min=0

dynamic.11.name=net_send
dynamic.11.source=speedtest-cli --simple --no-download | grep Upload | cut -d" " -f2
dynamic.11.regexp=(.*)
dynamic.11.postprocess=$1*131*8
dynamic.11.rrd=GAUGE
dynamic.11.interval=6
dynamic.11.min=0

web.status.1.content.8.name=Internet
web.status.1.content.8.icon=network.png
web.status.1.content.8.line.1="Public IP: " + InsertHTML("http://ifconfig.me/ip")
web.status.1.content.8.line.2='Upload: ' + KMG(data.net_send, 'k') + ' Download: ' + KMG(data.net_received, 'k')

web.statistics.1.content.2.name=Internet
web.statistics.1.content.2.graph.1=net_send
web.statistics.1.content.2.graph.2=net_received
web.statistics.1.content.2.ds_graph_options.net_send.label=Upload (Kbits/s)
web.statistics.1.content.2.ds_graph_options.net_send.lines={ fill: true }
web.statistics.1.content.2.ds_graph_options.net_send.color="#FF7777"
web.statistics.1.content.2.ds_graph_options.net_received.label=Download (Kbits/s)
web.statistics.1.content.2.ds_graph_options.net_received.lines={ fill: true }
web.statistics.1.content.2.7ds_graph_options.net_received.color="#77FF77"
```

- /etc/rpimonitor/daemon.conf (because speedtest-cli take about 30 sec to finish)

```
daemon.timeout=50
```

- The statistics page

[image]

Tell me if it's okay for you too :).

IssueFindings, 2020-01-27:
Hello @Leepic,

I just read now your solution. Really, many thanks for your help.
To well understand changes, could you explain me these lines :

- `dynamic.11.postprocess=$1*131*8`, what means these numbers ?

- What the difference between `grep Download | cut -d" " -f2` and `awk '/Download/ { print $2 }'` ? The last one is faster ? Why ?

For me, datas displayed with `web.status.1.content.8` are different than datas displayed with `web.statistics.1.content.2`. I mean, I have 298.04MB for upload status and 793,65Kbits/s for upload statistics. Why ?

I will keep interval=6 because I need to cheek my internet operator very well (I have doubts about the quality). Do you think that is possible to cheek also the real internet activities. I mean not the maximum upload and download speed, but the upload and download bandwidth really used by computer. Do you have any idea for that ?

Have a nice day.

IssueFindings, 2020-04-27:
Hello,
I find something. The issue is the unit used (keep in mind that speedtest-cli gives a result in Mbit/s). I propose the script below. That works perfectly. 

Have a nice day.

> dynamic.10.name=net_received
dynamic.10.source=speedtest-cli --simple --no-upload | awk '/Download/ { print $2 }'
dynamic.10.regexp=(.*)
dynamic.10.postprocess=
dynamic.10.rrd=GAUGE
dynamic.10.interval=6
dynamic.10.min=0
dynamic.11.name=net_send
dynamic.11.source=speedtest-cli --simple --no-download | awk '/Upload/ { print $2 }'
dynamic.11.regexp=(.*)
dynamic.11.postprocess=
dynamic.11.rrd=GAUGE
dynamic.11.interval=6
dynamic.11.min=0
web.status.1.content.8.name=Network
web.status.1.content.8.icon=network.png
web.status.1.content.8.line.1="IP: <b>" + InsertHTML("http://ifconfig.me/ip") + "</b>"
web.status.1.content.8.line.2="Upload: <b>"+ data.net_send+"Mbit/s</b> Download: <b>"+ data.net_received+"Mbit/s</b>"
web.statistics.1.content.2.name=Network
web.statistics.1.content.2.graph.1=net_send
web.statistics.1.content.2.graph.2=net_received
web.statistics.1.content.2.graph_options.yaxis={ tickFormatter: function (v) { if (Math.abs(v) > 1024) return (Math.round(v*10/1024)/10) + " Gbit/s" ; else return v + " Mbit/s" }, }
web.statistics.1.content.2.ds_graph_options.net_send.label=Upload
web.statistics.1.content.2.ds_graph_options.net_send.lines={ fill: true }
web.statistics.1.content.2.ds_graph_options.net_send.color="#FF7777"
web.statistics.1.content.2.ds_graph_options.net_received.label=Download
web.statistics.1.content.2.ds_graph_options.net_received.lines={ fill: true }
web.statistics.1.content.2.7ds_graph_options.net_received.color="#77FF77"

closed 2020-04-27 by IssueFindings