RPi-Monitor/RPi-Monitor #279: Display my public IP address

IssueFindings, 2019-07-19:
Hello,
I would like to display my public address directly in the webpage, without any variable created before. I think it's in the part `web.status.1.content.8.line.X`.
I write this line _web.status.1.content.8.line.1="IP : <b>" curl ifconfig.me "<i class='icon-arrow-up'></i></b>"_, but the result is always _-> SyntaxError: unexpected token: identifier_
Please, Where is my error ? Have a nice day.

Leepic, 2019-07-20:
Hello,

You need to use the insertHTML function, like this:

`web.status.1.content.8.line.4="IP: " + InsertHTML("http://ifconfig.me/ip")`

Restart rpimonitor and you public IP will be displayed.

Have a nice day.

IssueFindings, 2019-07-20:
Hello,

That works very fine ! Many thanks for your help.
Just a question, if the public IP change when RPI-Monitor will display it ? On real-time or with a delay ? How much ?

Have a nice day.

Leepic, 2019-07-20:
It will be displayed in real time :).

Have a nice day.

IssueFindings, 2019-07-21:
Great !! Many thanks for your help.

closed 2019-07-21 by IssueFindings

wrapper, 2020-11-28:
> `web.status.1.content.8.line.4="IP: " + InsertHTML("http://ifconfig.me/ip")`

Any idea how to show the IP address of the Pi and not that of the visitor of the page? Thanks

Leepic, 2020-11-29:
> > `web.status.1.content.8.line.4="IP: " + InsertHTML("http://ifconfig.me/ip")`
> 
> Any idea how to show the IP address of the Pi and not that of the visitor of the page? Thanks

The line above doesn't display the IP of the visitor page but his public IP address.

You can get the local IP of your RaspberryPi like this:

#### 1. Modify network.conf template ####

`sudo nano /etc/rpimonitor/template/network.conf`

```
static.1.name=local_IP
static.1.source=hostname --all-ip-addresses
static.1.regexp=(.*)
static.1.postprocess=

web.status.1.content.1.line.2='Local IP: <b>' + data.local_IP + '</b>'
```

**Note**: _network.conf_ need to be included in _/etc/rpimonitor/data.conf_ and _static_ means that the command is only executed at rpimonitor startup.

#### 2. Apply changes #### 

`sudo systemctl restart rpimonitor`

#### 3. Result #### 

[image]

**Note**: If you have an IPv6, it will be display on the same line too.

wrapper, 2021-05-02:
> **Note**: _network.conf_ need to be included in _/etc/rpimonitor/data.conf_ and _static_ means that the command is only executed at rpimonitor startup.
> #### 2. Apply changes
>
> `sudo systemctl restart [...]

Thanks for this. However what I'm looking to do is show the public IP of the raspberry pi. The first solution shows your public IP as the visitor of the page - not the IP of the Pi itself, and the second is the local IP not the public IP.

adi170-alt, 2022-04-04:
> Thanks for this. However what I'm looking to do is show the public IP of the raspberry pi. The first solution shows your public IP as the visitor of the page - not the IP of the Pi itself, and the [...]

This worked for me:
`static.2.name=public_IP
static.2.source=hostname -I | awk '{print $5}'
static.2.regexp=(.*)
static.2.postprocess=

web.status.1.content.8.line.2='Local IP: <b>' + data.public_IP + '</b>'`

if u get a different IP then u want, then just go to your terminal and try hostname -I | awk '{print $5}' but with different numbers and see what ip u want, so for example hostname -I | awk '{print $1}' (that is my local ip)