RPi-Monitor/RPi-Monitor #280: IPC::ShareLite store() error: Invalid argument at /usr/bin/rpimonitord line 840

MarcleGrand, 2019-07-21:
Hello Everyone
Had RPI monitor up and running and all working fine.
I use my RPI for ADS-B reception and ran into an issue on a RPI 3B+ with dump1090 v1.15 mutability. I used the https://forum.flightradar24.com/threads/10232-How-to-Install-dump1090-mutability_1-15-dev-on-RPi bug fix and ran the command sudo wget -O /etc/udev/rules.d/rtl-sdr.rules "https://raw.githubusercontent.com/osmocom/rtl-sdr/master/rtl-sdr.rules"  and sudo reboot which then stopped me from accessing the RPI Monitor. (not able to load in a browser my-RPI-IP:8888  Just says this website can't be reached.
I can reach all other website from my RPI. my-RPI-IP:8754, 10000. 30053 /graphs1090     /dump1090/gmap.html
rpimonitored and systemctl status rpimonitor result in the following:
[image]
Any ideas what I can do to make this work?
Thanks, Marc

Leepic, 2019-07-21:
Hello,

Just some questions:

- Which version of RPi-Monitor are you using? 
`rpimonitord -V`
- Is the port 8888 is in LISTEN state?
`ss -nat '( sport = :8888 )'`
- Is rpimonitord really running?
`ps -ef | grep rpimonitord`
- Can you try to launch rpimontord daemon manualy?
Kill it first:
`pkill rpimonitord`
Then launch it by hand:
` /usr/bin/perl /usr/bin/rpimonitord -b /var/run/rpimonitord.pid -l /var/log/rpimonitor.log`

MarcleGrand, 2019-07-21:
Thanks Leepic.
Hereby the results
[image]

Leepic, 2019-07-21:
The problem it's because RPi-Monitor is not running. The last command has failed because `/var/log/rpimonitor.log` need to be passed as an argument of `-l`, not on a new line but you can launch rpimonitord normally via systemd and tell me if it's working :).

`systemctl start rpimonitor`

MarcleGrand, 2019-07-21:
Tried `systemctl start rpimonitor` command. The command runs but didn't work (still not able to access the RPI Page via the browser my-RPI-IP:8888)

Leepic, 2019-07-21:
Hum, OK. RPi-Monitor is probably still not running.
So can you launch rpimonitord by hand:

- run as root user (eg. via sudo)

`sudo /usr/bin/perl /usr/bin/rpimonitord -b /var/run/rpimonitord.pid -l /var/log/rpimonitor.log`

Have you got any errors?

MarcleGrand, 2019-07-22:
See result below
[image]
No other errors as far as I know excet the one when using the `rpimonitored` resulting in IPC::ShareLite store() error: Invalid argument at /usr/bin/rpimonitord line 840
Happy to test anything else you can think off. I already tried to re-install with the same result.

Leepic, 2019-07-22:
Yes, your bash error is due to the wrong parameter on the command line (because /var/log/rpimonitor.log is not a command, but a file, so bash can execute it). Try this:

```sh
sudo /usr/bin/perl /usr/bin/rpimonitord -b /var/run/rpimonitord.pid -l /var/log/rpimonitor.log
```

Don't forget to put /var/log/rpimonitor.log at the **end** of the command, **not** on a newline.

MarcleGrand, 2019-07-22:
I found the problem
Doing `sudo nano /etc/rpimonitor/template/network.conf` and blocking (adding # infront) of a `netstat` for port 10000 made my RPI work again.
[image]
I don't think I made a mistake but correct me if I am wrong.
Can this be a bug in the RPI software?

Leepic, 2019-07-22:
Strange, I've tested and it works for me. If you uncomment it and restart rpimonitord, it's again not accessible?

Leepic, 2019-07-27:
If everything is works for you, you can close this issue :).

MarcleGrand, 2019-07-28:
Sorry for the late reply. The issue is still present/back again when I edit my `sudo nano /etc/rpimonitor/template/network.conf` and remove the hashes so I put them back.
[image]
The only thing I see now is that it will not show the CPU temp. I tried what is mentioned in [https://github.com/XavierBerger/RPi-Monitor/issues/245](url) but haven't succeded yet. Still trying

MarcleGrand, 2019-08-06:
The only way at the end was to delete and re-install from scratch.
Thanks

closed 2019-08-06 by MarcleGrand