RPi-Monitor/RPi-Monitor #282: Track running applications

IssueFindings, 2019-07-22:
Hello,
In addition with indicators about CPU and temperature, it could be interested to track running applications that consume the most resources. To do that, I think to use "top" to get the complete list but I need to sort results to keep only the top 5 and refresh each 10s.

Do you have any better idea ? And for the code ?

dynamic.1.name=running_applications
dynamic.1.source=top => I am not confortable with this part
dynamic.1.regexp=(.*)

web.status.1.content.1.title="The 5 most consuming apps"
web.status.1.content.1.icon=daemons.png
web.status.1.content.1.line.1= => I don't know how to display a dynamic table (refresh each 10s) with 5 lines (+ column headers)

Have a nice day.

Leepic, 2019-07-22:
Hello,

This feature already exists in RPi-Monitor, it's called the top3 addon (3 apps only but you can customize it, see below).
To enable it:

* Uncomment the line 69 in [/etc/rpimonitor/template/cpu.conf](https://github.com/XavierBerger/RPi-Monitor/blob/develop/src/etc/rpimonitor/template/cpu.conf#L69)
* Create the crontab
```
sudo crontab -e
* * * * * cd /usr/share/rpimonitor/web/addons/top3; ./top3 > top3.html)
* * * * * (sleep 10 ; cd /usr/share/rpimonitor/web/addons/top3; ./top3 > top3.html)
* * * * * (sleep 20 ; cd /usr/share/rpimonitor/web/addons/top3; ./top3 > top3.html)
* * * * * (sleep 30 ; cd /usr/share/rpimonitor/web/addons/top3; ./top3 > top3.html)
* * * * * (sleep 40 ; cd /usr/share/rpimonitor/web/addons/top3; ./top3 > top3.html)
* * * * * (sleep 50 ; cd /usr/share/rpimonitor/web/addons/top3; ./top3 > top3.html)
```
* Modify top3 addon
In [/usr/share/rpimonitor/web/addons/top3/top3
](https://github.com/XavierBerger/RPi-Monitor/blob/develop/src/usr/share/rpimonitor/web/addons/top3/top3#L22) replace this:
`my $nbtop=3;`
by
`my $nbtop=5;`
* restart rpimonitor
`systemctl restart rpimonitor`

Result under CPU (don't worry about the undefined values, it's because It's running in a container):

[image]
)

You can customize it if you want more/less informations.

IssueFindings, 2019-07-23:
Hello,
That works very well !! I didn't see this addon on RPI-Monitor documentation. I probably missed it. Sorry for this question.
It's necessary to add `* * * * * **root** cd /usr/share/rpimonitor/web/addons/top3; ./top3 > top3.html)` :-)

Just a question, what's the interest to put `* * * * * (sleep XX ; cd /usr/share/rpimonitor/web/addons/top3; ./top3 > top3.html)` ?

Have a nice day.

Leepic, 2019-07-27:
The documentation is [here](https://xavierberger.github.io/RPi-Monitor-docs/25_addons.html?highlight=top3#top3-addon).

It was needed because in your initial question your need was to refresh the page every 10 seconds and cron can't do it natively without this workaround ([source](https://stackoverflow.com/a/43066502)).

If everything is work for you, you can close this case :).

Have a nice day too.

IssueFindings, 2019-07-28:
Many thanks for these explanations !!

closed 2019-07-28 by IssueFindings