RPi-Monitor/RPi-Monitor #287: Monitor only runs after fresh install

khilendel, 2019-08-12:
Hello,

I am using a Raspberry PI Model 3 B (non-plus) with a relatively slow SD card (not sure if that matters)

I have tried both the manual install and repo install versions of Rpimonitor

The app/service only seem to start on fresh install. If i uninstall Rpimonitor and install it fresh, it starts up and I can access the web UI, after a reboot or a service restart, the service goes to "active (exited)" and I do not see any of the perl instances I normally see in Service status.

When I run "/usr/bin/rpimonitord -vvvvv" I get "IPC::ShareLite store() error: Invalid argument at /usr/bin/rpimonitord line 840.
rpimonitord stopped because respawning too fast."

Anything I can try?

After initial install (or fresh uninstall/reinstall):
[image]

After reboot or service restart:
[image]

Console output of "rpimonitord -vvvvv"
[image]

Trotter73, 2019-08-12:
Seeing the same here but on a ZeroW, as soon as I ran sudo /etc/init.d/rpimonitor update the problems started, the service starts fine with sudo /etc/init.d/rpimonitor start

khilendel, 2019-08-13:
Also running Buster. Additionally I have a Pi4 on buster, installed same way (using repo) that works fine.

Trotter73, 2019-08-13:
I’ve done a bit more investigation, upon reboot I see the following in rpimonitor.log

[2019/08/13-11:28:55] RPi-Monitor has been started
                      Log level can be increased or decreased during runtime
                      by sending signal USR1 or USR2 with the command killall -USRx rpimonitord
Web server not started because of error: Invalid argument
rpimonitord stopped because respawning too fast.

sudo service rpimonitor start does not start the service I have to use sudo service rpimonitor restart to be able to get into the web interface.

jwtaylor310, 2019-08-13:
I have run into the same problem with RPi-Monitor on a Pi running version 3.2.0 of NOOBS.  I reinstalled Raspbian using NOOBS version 2.8.2 on the same hardware and with that configuration RPi-Monitor runs without a problem.  I've done several "clean" installs of both versions of Raspbian and the pattern is consistent, so it looks like the cause is a conflict between RPi-Monitor and version 3.x of Raspian.

The log error that I am seeing is:

IPC::ShareLite store() error: Invalid argument at /usr/bin/rpimonitord line 822.
rpimonitord stopped because respawning too fast.

Line 822 of this instance of rpimonitord is:

$configuration->{'sharedmem'}->store( $json );

khilendel, 2019-08-15:
Weirdly, mine is the same line of code ($configuration->{'sharedmem'}->store( $json );) but called on out line 840

jwtaylor310, 2019-08-19:
I have done another "clean" install of Raspbian and RPi-Monitor and now have a different issue.  I am no longer getting the "IPC::ShareLite store" error.  Instead, when the Pi is booted the log shows "Web server not started because of error:  Invalid argument".  If I restart RPi-Monitor (using sudo /etc/init.d/rpimonitor restart), everything loads and runs normally.  I reconfigured the system so RPi-Monitor  does not start at boot, and when I manually start it (using sudo /etc/init.d/rpimonitor start) it starts without a problem.  So, it appears that at boot RPi-Monitor is trying to start before something in the environment is fully initialized.  I have done repeated clean installs of Raspbian and RPi-Monitor with consistent results.  The problem seems to be associated with the "Buster" version of Raspbian which is included in version 3 of NOOBS.  The problem does not occur with the earlier "Stretch" version included in version 2 of NOOBS.  The "IPC::ShareLite store" error has not recurred, possibly due to an update to Raspbian.

jwtaylor310, 2019-08-19:
...Also, the "respawning too fast" error message and exit is probably a symptom rather than a cause.  It looks like the main program has a loop that checks the elapsed time on each pass through the loop.  When the setup dies the time in the loop is shortened, resulting in the "respawning too fast" error.  That error goes away when everything initializes correctly.

frederick0102, 2019-08-19:
This error occures when I updated from Scretch to Buster. 
My var/log/rpimonitor.log contains exactly what you mentioned:
"Web server not started because of error: Invalid argument
rpimonitord stopped because respawning too fast."

Can anyone advise something how to hotfix this?

Thank you.

khilendel, 2019-08-20:
Jwtaylor - my results mirror yours - after running my apt updates today I am able to restart rpimonitor and it comes up. 

Previously restarting would put it back in to active(exited) state, but as of today's updates applied if I restart the service from init.d or the service command it comes up fine.

Is there a way to institute a way of delaying a service start, like in windows? or should I just trust this behavior will likely improve in subsequent buster releases?

mohawk-c, 2019-08-20:
Same after last apt upgrade.
If started by init.d script same error:
"Web server not started because of error: Invalid argument
rpimonitord stopped because respawning too fast."

If run from command line (rpimonitord without any params) - works...

jwtaylor310, 2019-08-21:
I have had success working around this problem by adding a time delay to the "Start" routine in rpimonitor.  On my Pi the required delay is about 7 seconds for a boot from power off, and less for a "warm" reboot.  I am currently using a 10 second delay to be on the safe side.  This is with a clean install of  Pi 3b using NOOBS v3.2.0 and the current release of RPi-monitor (2.12-r0).  If your hardware runs at a different speed or you are using a different version of the software, you may need to increase the delay.  

To add the delay to the startup script, edit rpimonitor (sudo nano /etc/init.d/rpimonitor).   Add the instruction "sleep 10" immediately before the "start" instruction in the case structure at tne end of the program.  This will be immediately after line 93 in the current release.  After editing, the section should look like this:

case "$1" in
    start)
      sleep 7
      start

Save the edited file and RPi-monitor should start without a problem on reboots and restarts.

LIMITATIONS:

1. If you reinstall RPi-monitor (or possibly following updates to the program) the edited rpimonitor script will be overwritten.  In that event, you will need to add the "sleep 10" instruction to the new version of the script.

2. The delay applies whether RPi-monitor is started automatically or manually.  I am looking into code that would skip the delay when the script is run manually and will post anything useful here.

mohawk-c, 2019-08-21:
@jwtaylor310 looks that your solution works .... Thanks!!!

Trotter73, 2019-08-22:
@jwtaylor310 Perfect thanks for sharing, really appreciated !

jwtaylor310, 2019-08-24:
I've been unsuccessful in trying to find a way to have a delay on automatic startup but no delay when RPi-monitor is started manually.  If the manual start delay annoys you, I have come up with the following work-around:

Add a new  option (I have used "startnow") which bypasses the delay.  Automatic starts will continue to use the "start" option but when you start manually you can use either "rpimonitor start" to start with the delay or "rpimonitor startnow" to start without the delay.  With this change the code starting at line 93 of /etc/init.d/rpimonitor becomes:

<pre>
    start)
        sleep 10
        start
    ;;
    startnow)
        start
    ;;
</pre>

sivelasco, 2020-04-05:
Seems to be fixed with #293

closed 2020-06-01 by XavierBerger