RPi-Monitor/RPi-Monitor #288: Monitoring External HDD's

dazbobaby, 2019-08-17:
I've struggled to get this working, then I looked at the code sdcard.conf and copied the requirements to storage.conf

In sdcard.conf you put:
```
static.7.name=sdcard_root_total
static.7.source=df /
static.7.regexp=\S+\s+(\d+).*\/$
static.7.postprocess=$1/1024
```
But in storage you've put:
```
static.10.name=storage1_total
static.10.source=df -t sda1
static.10.regexp=/sda1\s+(\d+)
static.10.postprocess=$1/1024
```
This failed for me everytime, I've changed it too:
```
static.10.name=storage1_total
static.10.source=df /dev/sda1
static.10.regexp=/sda1\s+(\d+)
static.10.postprocess=$1/1024
```

So I removed the -t and put /dev/sda1 and this worked for me.

Now if I could do this with a network share.
I've tried /mnt/samba and no luck
I've tried //192.168.1.1/H and no luck
I've tried /192.168.1.1/H - no luck.

Though df -h /mnt/samba works on the command line, as does df -h //192.168.1.1/H, it doesn't work for rpimonitor

Excellent work though, I love it, my thanks to you for making it.

dazbobaby, 2019-08-28:
I have my NAS connected though NFS and NFS can work too.

static.11.name=storage2_total
static.11.source=df /mnt/Movies
static.11.regexp=/Movies\s+(\d+)
static.11.postprocess=$1/1024

dynamic.15.name=storage2_used
dynamic.15.source=df /mnt/Movies
dynamic.15.regexp=/Movies\s+\d+\s+(\d+)
dynamic.15.postprocess=$1/1024
dynamic.15.rrd=GAUGE

web.status.1.content.9.name=Storage
web.status.1.content.9.icon=usb_hdd.png
web.status.1.content.9.line.1="<b>/500GB HDD</b> Used: <b>"+KMG(data.storage1_used,'M')+"</b> (<b>"+Percent(data.storage1_used,data.storage1_total,'M')+"</b>) Free: <b>"+KMG(data.storage1_total-data.storage1_used,'M')+ "</b> Total: <b>"+ KMG(data.storage1_total,'M') +"</b>"
web.status.1.content.9.line.2=ProgressBar(data.storage1_used,data.storage1_total)
web.status.1.content.9.line.3="<b>Movies</b> Used: <b>"+KMG(data.storage2_used,'M')+"</b> (<b>"+Percent(data.storage2_used,data.storage2_total,'M')+"</b>) Free: <b>"+KMG(data.storage2_total-data.storage2_used,'M')+ "</b> Total: <b>"+ KMG(data.storage2_total,'M') +"</b>"
web.status.1.content.9.line.4=ProgressBar(data.storage2_used,data.storage2_total)

Conver, 2019-08-28:
I need this too, I need to be able to monitor the NAS connected to the pi running RPi-monitor

GhislainOlitec, 2019-10-26:
It does not work for me

`df th /mnt/nas1Sys. de fichiers blocs de 1K   Utilisé Disponible Uti% Monté sur
/dev/sdb1          479667880 131200092  324032232  29% /mnt/nas1
`

GhislainOlitec, 2019-10-26:
It don't work for me, I have a "NaNkB" that appears....

If i type
`df /mnt/nas1`

he returns 
`Sys. de fichiers blocs de 1K   Utilisé Disponible Uti% Monté sur`
`/dev/sdb1          479667880 131200092  324032232  29% /mnt/nas1`

so if I understand well as the "/nas1" is in the end he is not looking forward ?? I also tried with "/dev" but don't work either....

I do not know REGEX so it's difficult for me. I will deepen the thing and come back if I have a solution ... because the problem is that each reboot sdX changes and the statistics too ...

`static.20.name=nas1_total`
`static.20.source=df /mnt/nas1`
`static.20.regexp=/nas1\s+(\d+)`
`static.20.postprocess=$1/1024`

`dynamic.21.name=nas1_used`
`dynamic.21.source=df /mnt/nas1`
`dynamic.21.regexp=/nas1\s+\d+\s+(\d+)`
`dynamic.21.postprocess=$1/1024`
`dynamic.21.rrd=GAUGE`

`web.status.1.content.9.line.3="<b>/Nas1</b> Used: <b>"+KMG(data.nas1_used,'M')+"</b> (<b>"+Percent(data.nas1_used,data.nas1_total,'M')+"</b>) Free: <b>"+KMG(data.nas1_total-data.nas1_used,'M')+ "</b> Total: <b>"+ KMG(data.nas1_total,'M') +"</b>"`
`web.status.1.content.9.line.4=ProgressBar(data.nas1_used,data.nas1_total)`

closed 2020-06-01 by XavierBerger