RPi-Monitor/RPi-Monitor #290: Pi4 Voltage undefinedV

blibenson, 2019-09-02:
Don't suppose anyone has come up for a fix for this yet?

[image]

GhislainOlitec, 2019-10-27:
Does the command work? because it can come from vcgencmd

`vcgencmd measure_volts core`

arvdrpoo, 2020-03-02:
I have noticed this as well. 
I think it may have something to do with the fact there are 3 terms in the command.

fernandofussuma, 2020-08-17:
Edit the file `/etc/rpimonitor/template/cpu.conf`;
```
-dynamic.2.source=vcgencmd measure_volts core
+dynamic.2.source=/opt/vc/bin/vcgencmd measure_volts core
```
restart the service or reboot:
`systemctl restart rpimonitor`

DaniW42, 2020-09-10:
Can somebody confirm that this issue isn't a misconfiguration. latest version works for me (Raspbian GNU/Linux 10 (buster)).
Could there be something wrong with the PATH environmental variable?

@blibenson What does `echo $PATH` show?

Marcoblue, 2020-09-24:
I installed thr RPI_Monitor yesterday on a fresh Pi 3B+. No problem with this issue.

alexgeek, 2022-06-22:
Grave digging but likely the same issue as me? The user needs to be in video group.

https://chewett.co.uk/blog/258/vchi-initialization-failed-raspberry-pi-fixed/