RPi-Monitor/RPi-Monitor #294: repository is down

mushu999, 2019-10-26:
pi@pihole:~ $ sudo apt update && sudo apt upgrade -y && exit
Get:1 http://archive.raspberrypi.org/debian buster InRelease [25.2 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Ign:3 http://giteduberger.fr rpimonitor/ InRelease
Err:4 http://giteduberger.fr rpimonitor/ Release
  Redirection loop encountered
Reading package lists... Done
E: The repository 'http://giteduberger.fr rpimonitor/ Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
pi@pihole:~ $

XavierBerger, 2019-10-30:
Server is working and has no reveal any problem.

Closing this issue.

closed 2019-10-30 by XavierBerger

guillaumedavidphd, 2019-11-20:
Same problem here. Same output. I followed the instructions in the documentation down to the T.
Suggestions?

guillaumedavidphd, 2019-11-20:
Solved, check out #267.