RPi-Monitor/RPi-Monitor #296: Struggling with a custom data collection frequency.

journeym, 2019-11-06:
I cant understand how should i create custom data collection delay for my data type.
Documentation states it is possible with custom rrd, but how shoul i implement that?
Does that mean that i should use rrdtool to create custom rrd file? 
How then im supposed to link that file with data type?

journeym, 2019-12-14:
Please help with that

Leepic, 2019-12-25:
Can you give me an example of what you want to do?

You want to extract dynamic data with custom interval (cf. [doc](https://xavierberger.github.io/RPi-Monitor-docs/27_configuration_templates.html#interval))?

journeym, 2019-12-26:
i want to get smart data once every 12h
For obvious reasons i do not want to set global dynamic data collection
delay, affecting all dynamic data types

Leepic, 2019-12-28:
> i want to get smart data once every 12h For obvious reasons i do not want to set global dynamic data collection delay, affecting all dynamic data types

Ok, so use the following configuration:

```
dynamic.1.name=interval1
dynamic.1.source=date
dynamic.1.regexp=(.*)

dynamic.2.name=interval2
dynamic.2.source=date
dynamic.2.regexp=(.*)
dynamic.2.interval=4320

web.status.1.content.1.line.1='Interval=10s last update at <b>'+data.interval1+'</b>'
web.status.1.content.1.line.2='Interval=12h last update at <b>'+data.interval2+'</b>'
```

The output result will be like this for example:

```
Interval=10s last update at Sat Dec 28 06:09:07 UTC 2019
Interval=12h last update at Sat Dec 28 18:09:07 UTC 2019
```

By default, the data extraction is every 10 secondes, so with a custom interval of 4320, the item's data **interval2** will be extracted only every **12 hours** (**4230***10=42300 secondes). interval1 will be extracted every 10 secondes here (no interval is the same as `dynamic.1.interval=1`).

The default extraction delay is configured by default in this file:

* /etc/rpimonitor/daemon.conf

``` 
daemon.delay=10
```

Be careful by modifying this globaly, it will affect all your dynamic item default interval and all you're history will be lost (RRD files will be recreated).

Does that satisfy your need?

journeym, 2019-12-28:
Shit, that was obvious

closed 2019-12-28 by journeym