RPi-Monitor/RPi-Monitor #299: After few hours i can't access rpimonitor via http://raspberrypi.local:8888

zblast5000, 2019-11-30:
After few hours i can't access rpimonitor via  http://raspberrypi.local:8888, when i try run rpimonitord i receive the message:
[2019/11/30-08:10:39] RPi-Monitor has been started
Log level can be increased or decreased during runtime
by sending signal USR1 or USR2 with the command killall -USRx rpimonitord
Web server not started because of error: Address already in use
rpimonitord stopped because respawning too fast.

brantje, 2019-12-05:
Same here with the docker image. Sometimes i just get a gateway timeout (using reverse proxy).

uksbi, 2020-04-23:
Same here in raspberry light (FR24 image). It looks like that it's happened after 8 hours.

Leepic, 2020-05-03:
Can you tell me if you have the errors describe in #218 ?

XavierBerger, 2020-06-01:
no answer, closing

closed 2020-06-01 by XavierBerger