RPi-Monitor/RPi-Monitor #301: addon top3 - web status display shows "undefined"

raz75, 2019-12-12:
I have modified the data.conf file to enable the top3 add on. It seems to work fine in addon web link. However on the status page, I see the "undefined" title on top of "top3" status. The status is showing correctly the top three processes. however I cannot seem to get rid of "undefined" string on the status  page.

Relevant lines from data.conf are:

web.status.1.content.1.line.1=InsertHTML("/addons/top3/top3.html")
web.addons.2.name=Top3 Processes --> I did this on purpose and it works fine on Addon page
web.addons.2.addons=top3

Leepic, 2019-12-25:
Have you follow [the steps](https://xavierberger.github.io/RPi-Monitor-docs/25_addons.html#top3-addon) decribe in the doc?

Maybe you have missing to enable the cron task and so there is no value to display.

pedrof1gueiredo, 2020-01-19:
I have the same issue. There is a old case that suggest changing the web add-on id number, but that does not fix this issue.

The add-on itself works and refreshes, but it has the "undefined" string at the top and is positioned in the first position of the monitor window.

Any suggestion?

etfovac, 2020-04-30:
> Have you follow [the steps](https://xavierberger.github.io/RPi-Monitor-docs/25_addons.html#top3-addon) decribe in the doc?
> 
> Maybe you have missing to enable the cron task and so there is no value to display.

What are the exact steps to set up the top3 addon? Where is the top3.html that displays the table? The top3.html in addon folder displays only addon info.

mattleebrown, 2020-05-02:
I tried adding the name and an icon.

web.status.1.content.1.name=Top3                                                                                        web.status.1.content.1.icon=top3.png

Worked for me!

closed 2020-06-01 by XavierBerger

Manevitch, 2022-09-26:
> web.status.1.content.1.line.1=InsertHTML("/addons/top3/top3.html")

This is a super-stale thread, but...

You can just put that line (or just uncomment the one that's already there) in the cpu.conf file rather than try to make Top3 a standalone section.