RPi-Monitor/RPi-Monitor #302: RPI monitor does not restart after reboot in Raspian Buster (Debian 10)

bartoosthoek, 2019-12-25:
I have installed RPI-monitor on a fresh Raspian Buster installation. All works fine except that Rpi-monitor deamon does not become active after reboot or start-up.  

running "/etc/init.d/rpimonitor status" shows below result:
__________
rpimonitor.service - LSB: RPi-Monitor daemon
   Loaded: loaded (/etc/init.d/rpimonitor; generated)
   Active: active (exited) since Wed 2019-12-25 13:02:19 CET; 6min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 328 ExecStart=/etc/init.d/rpimonitor start (code=exited, status=0/SUCCESS)

Dec 25 13:02:16 b..r-new systemd[1]: Starting LSB: RPi-Monitor daemon...
Dec 25 13:02:19 b..r-new rpimonitor[328]: Starting RPi-Monitor:.
Dec 25 13:02:19 b..r-new systemd[1]: Started LSB: RPi-Monitor daemon. 
_________

After running "/etc/init.d/rpimonitor restart" all works perfect and shows status as below

_________
 rpimonitor.service - LSB: RPi-Monitor daemon
   Loaded: loaded (/etc/init.d/rpimonitor; generated)
   Active: active (running) since Wed 2019-12-25 13:01:04 CET; 35s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 1742 ExecStart=/etc/init.d/rpimonitor start (code=exited, status=0/SUCCESS)
    Tasks: 3 (limit: 2200)
   Memory: 12.9M
   CGroup: /system.slice/rpimonitor.service
           ├─1750 /usr/bin/perl /usr/bin/rpimonitord -b /var/run/rpimonitord.pi…
           ├─1767 /usr/bin/perl /usr/bin/rpimonitord -b /var/run/rpimonitord.pi…
           └─1768 /usr/bin/perl /usr/bin/rpimonitord -b /var/run/rpimonitord.pi…

Dec 25 13:01:03 b..r-new systemd[1]: Starting LSB: RPi-Monitor daemon...
Dec 25 13:01:04 b..r-new rpimonitor[1742]: Starting RPi-Monitor:.
Dec 25 13:01:04 b..r-new systemd[1]: Started LSB: RPi-Monitor daemon. 
________

What could this be? Installing on Raspian Stretch (Debian 9) shows no issues. Currently I am running a cron job which waits for 10 seconds and than restarts Rpi, however this is a far from pretty solution. 

Thnx in advance

Leepic, 2019-12-25:
Hello @bartoosthoek ,

Maybe it's related to the PR #293 ?

Try this in `/etc/init.d/rpimonitor`:

```diff
--# Required-Start:    $remote_fs $syslog
--# Required-Stop:     $remote_fs $syslog
++# Required-Start:    $remote_fs $syslog $network
++# Required-Stop:     $remote_fs $syslog $network
```

bartoosthoek, 2019-12-26:
Hi @Leepic , thnx for the quick reply. I changed the "rpimonitor" script, but unfortunately it did not resolve the issue for me. 

As I am not a expert, I only just need to change the file correct? Or did I forgot something?

Thnx again

Leepic, 2019-12-28:
>As I am not a expert, I only just need to change the file correct? Or did I forgot something?

Yes, you're right.

Revert the two modifications and then can you give me the output of thoses commands:

* Display details about rpimonitor package

```shell
apt-cache showpkg rpimonitor
```

* When the problem occurs (display informations about listing files that match port 8888)

```shell
sudo lsof -i :8888 -n
``` 

* Do you use RPi-Monitor behind a reverse proxy or the embedded web server inside RPi-Monitor?
* It's working if you only start (not restart) rpimonitor (`/etc/init.d/rpimonitor start`)?

bartoosthoek, 2019-12-28:
Hi @Leepic, 

Thanks again for your support, please see the replies on your questions below:

>     * Display details about rpimonitor package
> ```shell
> apt-cache showpkg rpimonitor
> ```

```shell
Package: rpimonitor
Versions:
2.12-r0 (/var/lib/apt/lists/giteduberger.fr_rpimonitor_Packages) (/var/lib/dpkg/status)
 Description Language:
                 File: /var/lib/apt/lists/giteduberger.fr_rpimonitor_Packages
                  MD5: adde0f330c2e7e6ce46cb12f9157aab6

Reverse Depends:
Dependencies:
2.12-r0 - perl (0 (null)) librrds-perl (0 (null)) libhttp-daemon-perl (18 6.0.0) libwww-perl (3 6.0.0) libjson-perl (0 (null)) libipc-sharelite-perl (0 (null)) libfile-which-perl (0 (null)) aptitude (0 (null))
Provides:
2.12-r0 -
Reverse Provides:
```

>     * When the problem occurs (display informations about listing files that match port 8888)
> 
> 
> ```shell
> sudo lsof -i :8888 -n
> ```

- No return on the shell when problem occurs 
- Below return when I have restarted (working Rpi)
```shell
COMMAND    PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
rpimonito 1916   pi    4u  IPv4  23155      0t0  TCP *:8888 (LISTEN)
```

>     * Do you use RPi-Monitor behind a reverse proxy or the embedded web server inside RPi-Monitor?
- Nope

>     * It's working if you only start (not restart) rpimonitor (`/etc/init.d/rpimonitor start`)?
- No, only "restart" works. Proving only start will result in the same output on status

Regards and Thnx

Leepic, 2019-12-28:
>Hi @Leepic,
>Thanks again for your support, please see the replies on your questions below:

Hello @bartoosthoek ,

With pleasure ;).

I think that the problem is the conversion between sysV init script and systemd but I can't reproduce your issue in my lab, maybe because I'm not on Raspbian on a RaspberryPi.

Can you modify the following in `/etc/init.d/rpimonitor`:

```diff
++# pidfile: /var/run/rpimonitord.pid
--# Required-Start:    $remote_fs $syslog
--# Required-Stop:     $remote_fs $syslog
++# Required-Start:    $remote_fs $syslog $network
++# Required-Stop:     $remote_fs $syslog $network
```
Note: add the `pidfile:` line just between the sheban `#!/bin/sh` and `### BEGIN INIT INFO`

Then save and reboot.

Tell me if it works. Otherwise, can you send me the output of this command:

`systemctl cat rpimonitor`

It will output the generated systemd script, based on the sysV init script mentioned above.

dieter114, 2020-01-06:
Hi @Leepic,

This works fine!
Thank`s for your help and greetings from Germany
Wolfdieter

bartoosthoek, 2020-01-06:
Hi @Leepic , 

Thanks or the support, however I am still a bit stuck with some strange "random" behaviour. 

If I follow your instructions all works fine if I shut-down the pi and give it a power cycle (reboot does not seem to work). But when I repeat the shut-down it does not work anymore, except when I redo below commands:  

> `sudo systemctl daemon-reload`
> `sudo systemctl enable rpimonitor`

As it seems to work for other people, it might be another issue with my set-up. I will create a fresh image of Raspian Buster next weekend an try again, just to be sure I am not fighting another problem right now.

I'll keep you posted of the results

Thanks again for all the help and best wishes for 2020!

hardy-b, 2020-03-18:
Hi there,

I am currently working on a project using a pi zero w and a pi 4 (2 gig). Both are running the same raspbian buster lite based on the current image from 2020-02-13. The 4 is my development system and the zero is the test system, therefore they are set up indentically software-wise.
 
Now my zero w behaves exactly as described by @bartoosthoek while on the 4 rpi monitor is running normally after reboot.

As strange as it may sound, I could even reproduce it setting up the zero completely from scratch.

Just wanted to let you know, maybe this is of any help for you.

Regards from Germany

Hardy

PrunelliParis, 2020-03-18:
Hi all

Just to let you know that I experienced the same issue since I upgraded from Stretch to Buster on my RPI 3

Fix provided by @Leepic here above worked like a charm in my case

Hope it helps,

Best

Dawilli

htcfreek, 2020-03-20:
Hello.

I have two installations of rpimonitor:
1. Raspberry Pi (Model 2) with Raspian Buster (10)
2. Test-VM (Virtual Box) with Raspian Buster (10) x86

At the beginning rpimonitor works fine on both systems. But after reinstalling rpimonitor on my Raspberry Pi I had the same problem descibed by @bartoosthoek. 

The solution for me was the Fix provided by @Leepic : [See heere.](#issuecomment-569456631)
Thanks for that @Leepic ! You have save me much time of searching and fixing.

Bye.
htcfreek

meeki007, 2020-04-13:
@Leepic 

Thanks M8 your fix worked like a charm

XavierBerger, 2020-06-01:
Fixed : closing

closed 2020-06-01 by XavierBerger