RPi-Monitor/RPi-Monitor #305: Error on apt update:  Redirection loop encountered

git-developer, 2020-02-07:
It seems the HTTP server configuration was changed in an apt-incompatible way:

```
$ cat /etc/apt/sources.list.d/rpimonitor.list
deb http://giteduberger.fr rpimonitor/

$ apt update
Ign:3 http://giteduberger.fr rpimonitor/ InRelease
Err:4 http://giteduberger.fr rpimonitor/ Release
  Redirection loop encountered
Reading package lists... Done
E: The repository 'http://giteduberger.fr rpimonitor/ Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

$ curl -v http://giteduberger.fr/rpimonitor/Release
* Connected to giteduberger.fr (213.186.33.17) port 80 (#0)
> GET /rpimonitor/Release HTTP/1.1
> Host: giteduberger.fr
>
< HTTP/1.1 307 Temporary Redirect
< Location: http://giteduberger.fr/rpimonitor/Release?__r=1.6b5b9d708a1339545f5b10509d319937

$ curl -v http://giteduberger.fr/rpimonitor/Release?__r=1.6b5b9d708a1339545f5b10509d319937
[...]
< HTTP/1.1 307 Temporary Redirect
< Set-Cookie: __r=1.6b5b9d708a1339545f5b10509d319937; path=/; Max-Age=60;
< Location: http://giteduberger.fr/rpimonitor/Release
```

git-developer, 2020-02-07:
Fix according to #267 (and #294): change URL to https:
`deb https://giteduberger.fr rpimonitor/` instead of
`deb http://giteduberger.fr rpimonitor/`

git-developer, 2021-08-03:
The redirection loop is back in effect, for both http and https URLs.

git-developer, 2021-08-04:
The redirection loop does not occur anymore.

closed 2021-08-04 by git-developer

polarweasel, 2021-10-01:
The redirection loop is occurring again.