RPi-Monitor/RPi-Monitor #309: [visibility bug] HTML-Tag "<i class='icon-arrow-up'></i>"does not work

htcfreek, 2020-03-21:
Hello.

With the html tag "<i class='icon-arrow-up'></i>" you can show symbols in the text on the html page. But this tag won't work for me in rpimonitor 2.12-r0. I use the original text form the template "/etc/rpimonitor/template/network.conf".
I have seen this on Microsoft Edge 8Chromium) and Firefox.

Is this a browser bug or an rpimonitor bug?

Thanks for your help.
htcfreek

wrapper, 2020-11-28:
+1 I also don't see it

GlynHu, 2021-01-26:
Me three!

This affects both network.conf & wlan.conf

The stats and layout all look good but no up/down arrows.

[image]

The respective calls appear identical in both .conf files and looks like this rerspectively;

web.status.1.content.8.line.1="Ethernet Sent: <b>"+KMG(data.net_send)+"<i class='icon-arrow-up'></i></b> Received: <b>"+KMG(Math.abs(data.net_received)) + "<i class='icon-arrow-down'></i></b>"

web.status.1.content.9.line.1="WiFi Sent: <b>"+KMG(data.wifi_send)+"<i class='icon-arrow-up'></i></b> Received: <b>"+KMG(Math.abs(data.wifi_received)) + "<i class='icon-arrow-down'></i></b>"

Not sure where it might be looking for the arrows but they don't exist in the /web/img folder.

Obviously not mission critical but it is in there for a reason so should work and would look pretty if it did!

Again if there is anything I can do to help then please let me know.

Kind regards,
-=Glyn=-

Drarig29, 2021-11-29:
This is linked to this: https://stackoverflow.com/questions/25466519/bootstrap-3-2-0-glyphicons-not-working

This is the version used my RPi-Monitor.

sgrayban, 2021-12-24:
@Drarig29 awesome fix!

[image]

GlynHu, 2022-01-17:
> @Drarig29 awesome fix!
> 
> [image]

I like the Net Speed icon & layout @sgrayban - Can I ask how you did this please?

It's been a looong time since I installed, configured & played with RPIMonitor and I've forgotten more than I ever learned! :)

Thanks & kind regards,
-=Glyn=-

Drarig29, 2022-01-17:
Here is the list of glyphicons: https://getbootstrap.com/docs/3.3/components/

But the bigger icons like "Network" and "Net speed" do not come from this.

You'll need to find them by yourself and store them in `/usr/share/rpimonitor/web/img` I believe (according to [this folder](https://github.com/XavierBerger/RPi-Monitor/tree/develop/src/usr/share/rpimonitor/web/img)).

After this, you can refer to this images by their filename.

Example with [wifi.png](https://github.com/XavierBerger/RPi-Monitor/blob/develop/src/usr/share/rpimonitor/web/img/wifi.png):

https://github.com/XavierBerger/RPi-Monitor/blob/53a95542e13f0289e117da3d6a18559604b25c0f/src/etc/rpimonitor/template/wlan.conf#L23

sgrayban, 2022-01-18:
> > @Drarig29 awesome fix!
> > [image]
> 
> I like the Net Speed icon & layout @sgrayban - Can I ask how you did this please?
> 
> It's been a looong time since I installed, configured & played with RPIMonitor and I've forgotten more than I ever learned! :)
> 
> Thanks & kind regards, -=Glyn=-

Here is my icon
[image]

GlynHu, 2022-01-19:
Thanks both for taking the time to reply to my post.

I would also like to know the lines needed and how to call & display the Net Speed Ping, Download & Upload info if possible please? 

Thanks & kind regards,
-=Glyn=-

sgrayban, 2022-01-19:
See this post https://github.com/XavierBerger/RPi-Monitor/issues/383

GlynHu, 2022-01-24:
Great…thanks Scott I’ll give it a go.

 

Thanks & kind regards,

-=Glyn=-