RPi-Monitor/RPi-Monitor #311: Graphs not updating

Simon01011, 2020-03-30:
Hi!

I'm new to RPi-Monitor. But my problem is that some graphs do not show the latest data. CPU, temperature and uptime are okay. Memory, swap and all disk data show only graphs with old data. The data is shown up to the last restart of the rpimonitor. No data is lost, it is just not shown. The rrd files in /var/lib/rpimonitor/stat are all okay and have the same size.

What can I do about this problem? Thanks in advance!

meerweten, 2020-04-06:
Got a similar problem with my storage data.
seems to be data from around 12h old (missing 80GB of data according to rpimonitor)

mrblond18, 2020-05-09:
Guys, I encountered the same problem, with mem and swap graphs. It turns out that if the first line you define in a graph is a static value then the graph is not properly created. Both the mem and swap config file use the total amount of mem/swap as first line, those are static values. Just define this as last item and it works fine.

Same goes for storage data I guess, probably also there you use a static value in the graph.