RPi-Monitor/RPi-Monitor #312: Graphs not showing data, but data is shown on Status page

kjeldahl, 2020-04-01:
I have a problem with three dynamic measures I have added to the sdcard.conf template.

I have added the following entries to monitor specific disk usage of three directories:
```
dynamic.21.name=sdcard_unifi_backup_usage
dynamic.21.source=sudo du -s /var/lib/unifi/backup
dynamic.21.regexp=(\d+)
dynamic.21.postprocess=$1/1024
dynamic.21.rrd=GAUGE

dynamic.22.name=sdcard_unifi_db_usage
dynamic.22.source=sudo du -s /var/lib/unifi/db
dynamic.22.regexp=(\d+)
dynamic.22.postprocess=$1/1024
dynamic.22.rrd=GAUGE

dynamic.23.name=sdcard_mongodb_usage
dynamic.23.source=sudo du -s /var/lib/mongodb
dynamic.23.regexp=(\d+)
dynamic.23.postprocess=$1/1024
dynamic.23.rrd=GAUGE
```
Side question are the numbers following dynamic supposed to be unique just within single files or across all enabled templates? 

And I have added this to display them on the Status page. This works fine; progress bars are being displayed with correct values.
```
web.status.1.content.7.name=SD card
web.status.1.content.7.icon=sd.png
web.status.1.content.7.line.1="<b>/boot</b> Used: <b>"+KMG(data.sdcard_boot_used,'M')+"</b> (<b>"+Percent(data.sdcard_boot_used,data.sdcard_boot_total,'M')+"</b>) Free: <b>"+KMG(data.sdcard_boot_total-data.sdcard_boot_used,'M')+ "</b> Total: <b>"+ KMG(data.sdcard_boot_total,'M') +"</b>"
web.status.1.content.7.line.2=ProgressBar(data.sdcard_boot_used,data.sdcard_boot_total,60,80)
web.status.1.content.7.line.3="<b>/</b> Used: <b>"+KMG(data.sdcard_root_used,'M') + "</b> (<b>" + Percent(data.sdcard_root_used,data.sdcard_root_total,'M')+"</b>) Free: <b>"+KMG(data.sdcard_root_total-data.sdcard_root_used,'M')+ "</b> Total: <b>"+ KMG(data.sdcard_root_total,'M') + "</b>"
web.status.1.content.7.line.4=ProgressBar(data.sdcard_root_used,data.sdcard_root_total,60,80)
web.status.1.content.7.line.5="<b>/var/lib/unifi/backup</b> Used: <b>"+KMG(data.sdcard_unifi_backup_usage,'M') + "</b> (<b>" + Percent(data.sdcard_unifi_backup_usage,data.sdcard_root_total,'M')+"</b>)"
web.status.1.content.7.line.6=ProgressBar(data.sdcard_unifi_back_usage,data.sdcard_root_total,60,80)
web.status.1.content.7.line.7="<b>/var/lib/unifi/db</b> Used: <b>"+KMG(data.sdcard_unifi_db_usage,'M') + "</b> (<b>" + Percent(data.sdcard_unifi_db_usage,data.sdcard_root_total,'M')+"</b>)"
web.status.1.content.7.line.8=ProgressBar(data.sdcard_unifi_db_usage,data.sdcard_root_total,60,80)
web.status.1.content.7.line.9="<b>/var/lib/mongodb</b> Used: <b>"+KMG(data.sdcard_mongodb_usage,'M') + "</b> (<b>" + Percent(data.sdcard_mongodb_usage,data.sdcard_root_total,'M')+"</b>)"
web.status.1.content.7.line.10=ProgressBar(data.sdcard_mongodb_usage,data.sdcard_root_total,60,80)
```

On the graphs the three measures values are shown as being 0 all the time
```
web.statistics.1.content.4.name=Disks - root
web.statistics.1.content.4.graph.1=sdcard_root_total
web.statistics.1.content.4.graph.2=sdcard_root_used
web.statistics.1.content.4.graph.3=sdcard_unifi_backup_usage
web.statistics.1.content.4.graph.4=sdcard_unifi_db_usage
web.statistics.1.content.4.graph.5=sdcard_mongodb_usage
web.statistics.1.content.4.ds_graph_options.sdcard_root_total.label=Size of / (MB)
web.statistics.1.content.4.ds_graph_options.sdcard_root_total.color="#FF7777"
web.statistics.1.content.4.ds_graph_options.sdcard_root_used.label=Used on / (MB)
web.statistics.1.content.4.ds_graph_options.sdcard_root_used.color="#7777FF"
web.statistics.1.content.4.ds_graph_options.sdcard_unifi_backup_usage.label=Used on /var/lib/unifi/backup (MB)
web.statistics.1.content.4.ds_graph_options.sdcard_unifi_backup_usage.color="#77FF77"
web.statistics.1.content.4.ds_graph_options.sdcard_unifi_db_usage.label=Used on /var/lib/unifi/db (MB)
web.statistics.1.content.4.ds_graph_options.sdcard_unifi_db_usage.color="#FF77FF"
web.statistics.1.content.4.ds_graph_options.sdcard_mongodb_usage.label=Used on /var/lib/mongodb (MB)
web.statistics.1.content.4.ds_graph_options.sdcard_mongodb_usage.color="#FFFF77"
```

Any ideas? Or suggestions as to how I can debug this?

Simon01011, 2020-04-01:
Hi!

I have almost the same setup and no problems. The only difference to my setup seems to be that I don't use "sudo". It is not necessary as rpimonitor runs as root.

AFAIK, the "dynamic numbers" need to be unique for just a single file.

kjeldahl, 2020-04-01:
Hi @Simon01011 

I look around a bit and found that there are no .rrd files for my measures in /var/lib/rpimonitor/stat it only contains .rrd files for the builtin measures.

Do you have .rrd files for your custom measures?

Simon01011, 2020-04-01:
Yes I do! But you restarted after changing the config (as root)?
`/etc/init.d/rpimonitor restart`

Take a look at the log file "/var/log/rpimonitor.log". Increase the log level with "killall -USR1 rpimonitord", I used level 5. Is your custom source "sudo du -s /var/lib/unifi/backup" shown? All current values are shown as well in curly braces. What does it look like?

(Remember to decrease log level again to prevent a large log file!)

kjeldahl, 2020-04-01:
Thanks for the feedback.

It turned out to be a limitation of the name of the RRD data file. It must not exceed 19 chars, but there are no checks in place and the failure to create the RRD is not logged.

I have submitted a patch to the documentation.

closed 2020-04-01 by kjeldahl

Simon01011, 2020-04-01:
Glad to hear that you found this amazing cause for your error. Good job!

hajosattila, 2023-08-26:
> It is not necessary as rpimonitor runs as root.

I was wondering, how safe is it?