RPi-Monitor/RPi-Monitor #313: Cannot install: Redirection loop encountered

ptoews, 2020-04-01:
Hi,
I followed the instructions at https://xavierberger.github.io/RPi-Monitor-docs/11_installation.html but `sudo apt update` does not work:

```
pi@raspberrypi:~ $ sudo apt update
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Hit:3 http://raspbian.raspberrypi.org/raspbian buster InRelease
Ign:1 http://giteduberger.fr rpimonitor/ InRelease
Err:4 http://giteduberger.fr rpimonitor/ Release
  Redirection loop encountered
Reading package lists... Done
E: The repository 'http://giteduberger.fr rpimonitor/ Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

I tried changing the contents of the apt sources rpimonitor.list to the https variant as suggested in #305 and #267, but that didn't help:

```
pi@raspberrypi:~ $ sudo apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Ign:3 https://giteduberger.fr rpimonitor/ InRelease
Err:4 https://giteduberger.fr rpimonitor/ Release
  Redirection loop encountered
Reading package lists... Done
E: The repository 'https://giteduberger.fr rpimonitor/ Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

htcfreek, 2020-04-01:
Hi.

For me apt update with rpimonitor-source work.
But my machine only contact the following adresses:
`Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Hit:2 http://giteduberger.fr rpimonitor/ InRelease                       
Hit:3 http://archive.raspberrypi.org/debian buster InRelease             
Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Fetched 13.0 MB in 18s (714 kB/s)                                        
Reading package lists... Done
Building dependency tree       
Reading state information... Done
4 packages can be upgraded. Run 'apt list --upgradable' to see them.`

Have you tryed to delete and recreate `/etc/apt/sources.list.d/rpimonitor.list`?

ptoews, 2020-04-02:
Yes and now it works, strange! Didn't test again before replacing it though.

closed 2020-04-02 by ptoews