RPi-Monitor/RPi-Monitor #316: SNMP issues

racedowling, 2020-04-10:
I tried monitoring a cluster using ssh but the overhead was too high so now I'm trying to move toward snmp. In the documentation you mention /etc/snmp/snmpd.conf.rpimonitor which I found in the sources but also MIB-RPIMONITOR  which I cannot find. 
Also rpimonitord -m hangs up whenever I run it.

Leepic, 2020-05-09:
Which version of rpimonitord are you using?
`rpimonitord -V`

The [documentation is about](https://xavierberger.github.io/RPi-Monitor-docs/43_changelog.html) RPi-Monitor 2.13 which is not yet release, so some informations are wrong with the latest stable 2.12r0, that the case for SNMP.

What's the error when you're running `rpimonitord -m`?

racedowling, 2020-05-10:
Thanks for responding. I'm using 2.12. The -m flag is ignored and it tries to run another instance. I converted my entire configuration to use SNMP but many of the calls require a shell script on the target device, for example:

dynamic.102.source=snmpget -v1 -cpublic -Ovq clust-1 'nsExtendOutLine."voltage".1'   

calls a bash shell on clust-1 that runs vcgencmd. This is not ideal, which is why I want to try using rpimonitor as an SNMP agent. Even so it greatly reduced the overhead of using ssh to access each of the nodes. I'll download the beta and give it a try.

closed 2020-05-10 by racedowling