RPi-Monitor/RPi-Monitor #319: Updating to from v2.12 to v2.13-beta7

4k3or3et, 2020-04-21:
Hi guys

it has been a long time since last update on master branch. I guess the project is dead now so i would like to update my RPi Monitor from v2.12 to v2.13-beta7. 

Anybody could point me please how can i do that? Is it even possible?

arturnowok, 2020-04-26:
Anyone?

4k3or3et, 2020-04-26:
Anyone?

4k3or3et, 2020-05-08:
Please help guys...

Leepic, 2020-05-09:
Hi,

2.13-beta7 doesn't exists but 2.13-beta6 exists.

Regarding the [official documentation](https://xavierberger.github.io/RPi-Monitor-docs/12_custom_installation.html), you just need to do download and install the new debian package. I think that it's the same with a beta release.

Note that I do not test the below steps.

1. Stop RPi-Monitor

`sudo systemctl stop rpimonitor`

2. Copy the configuration or backup on another device (the 2nd is better)

```
mkdir /home/rpimonitor-2.12r0.bak
cp -a --parents /usr/share/rpimonitor /home/rpimonitor-2.12r0.bak
cp -a --parents /var/lib/rpimonitor /home/rpimonitor-2.12r0.bak
cp -a --parents /etc/rpimonitor /home/rpimonitor-2.12r0.bak
```

3. Download and install the latest dev release

```
wget https://github.com/XavierBerger/RPi-Monitor-deb/raw/develop/packages/rpimonitor_latest.deb
sudo dpkg -i rpimonitor_latest.deb
```
4. Start RPi-Monitor
`sudo systemctl start rpimonitor `
5. Check your version
`rpimonitord -V`
6. Test that nothing was broken during the upgrade
7. If everything is okay, comment the lines in `/etc/apt/sources.list.d/rpimonitor.list` to avoid future APT conflicts with your manually upgrade version.

XavierBerger, 2026-10-27:
Installation process has been redesigned.
See README.md for details.

closed 2026-10-27 by XavierBerger