RPi-Monitor/RPi-Monitor #320: No data in RPiMonitor from AM2302

Henk-Kuik, 2020-04-24:
I have a RaspberryPi Zero W with Raspbian Buster with desktop running on it.
Next to that I have a AM2302/DHT22 sensor with power on pin 2 (5 v), ground on pin 6 and data on pin 7 (GPIO 4). 

I installed RPiMonitor and did no configuration what so ever other then what follows.

In /etc/rpimonitor I only unchecked the line [include=/etc/rpimonitor/template/dht11.conf] in data.conf
So, I removed the hash.

I edited /etc/rpimonitor/template/dht11.conf to work with a am2302 on gpio pin 4.
Here is that file.
dht11.conf
-----------------------------
########################################################################
# Extract temperature and humidity from DHT11/DHT22 
#  Page: 2
#  Information               Status     Statistics
#  - tempetature             - yes      - yes
#  - humidity                - yes      - yes
#######################################################################
# Add new pages (number 2)
web.status.2.name=Living room
web.statistics.2.name=Living room
 
# Get dynamic data
dynamic.1.name=living_room_temp
dynamic.1.source=Adafruit_DHT 2302 4
dynamic.1.regexp=t=(\d+)$
dynamic.1.postprocess=$1/1000
dynamic.1.rrd=GAUGE

dynamic.2.name=humidity
dynamic.2.source=Adafruit_DHT 2302 4
dynamic.2.regexp=Hum = (\S+)
dynamic.2.postprocess=
dynamic.2.rrd=GAUGE

# Add a line of status in page number 2
web.status.2.content.1.name=Temperature
web.status.2.content.1.icon=cpu_temp.png
web.status.2.content.1.line.1="Ambient Temperature: <b>"+data.living_room_temp+"&deg;C</b>"
web.status.2.content.1.line.2="Ambient Humidity: <b>"+data.humidity+"%</b>"

# Add a statistic graph (with 2 curves suing separate scales) in page number 2
web.statistics.2.content.1.name=Temperature
web.statistics.2.content.1.graph.1=living_room_temp
web.statistics.2.content.1.graph.2=humidity
web.statistics.2.content.1.ds_graph_options.living_room_temp.label=Ambient temp. (deg C)
web.statistics.2.content.1.ds_graph_options.humidity.label=Humidity (%)
web.statistics.2.content.1.ds_graph_options.humidity.yaxis=2
web.statistics.2.content.1.graph_options.legend= { position: "sw"}
web.statistics.2.content.1.graph_options.y2axis={ position: "right", min: 0, max: 100 }
-----------------------------

Restarted RPiMonitor with [sudo service rpimonitor restart] but getting no data at all from the am2302 sensor.
Ambient Temperature: undefined°C
Ambient Humidity: undefined%

The AM2302/DHT22 sensor is working because when I do [sudo /home/pi/sources/Adafruit_Python_DHT/examples/AdafruitDHT.py 2302 4] I am getting output from this sensor.
For that I did install the Adafruit_Python_DHT sensor library obviously and I did this before installing RPiMonitor. 

In other words, I am missing or overlooking something but for the life of me I can not see what that is. Can someone push me in the right direction please? I would appreciate it very much.

Leepic, 2020-04-26:
Hello,

I think it's because your source is wrong. Adafruit_DHT is on your `$PATH`?
Replace ```source=Adafruit_DHT 2302 4```
by
```source=/home/pi/sources/Adafruit_Python_DHT/examples/AdafruitDHT.py  2302 4```

and maybe your regexp is wrong too because you need to get only number (% and °C), not string. Try this instead:

```
 dynamic.1.regexp=(\d+)$
```

``` 
 dynamic.2.regexp=(\d+)$
```

Henk-Kuik, 2020-04-29:
I have added [ /home/pi/sources/Adafruit_Python_DHT/examples ] to PATH and when I do [ echo $PATH ] it does show up.
In DHT11.conf I gave source the full path as you mentioned and I edited regexp also to what you mentioned but still I get no data.
All I get is Ambient Temperature: 0.00°C Ambient Humidity: 0.00%

Leepic, 2020-05-03:
Have you follow the steps describe in the doc?
https://xavierberger.github.io/RPi-Monitor-docs/32_sensors.html#software-installation

if you add sudo in the source, have you got values?

Henk-Kuik, 2020-05-22:
I am very sorry to inform you that the tutorial you mention is not working anymore. Immediately after step one I get a zero kB file back and a 404: Not Found error.

Leepic, 2020-08-02:
Yes, it's because the repository was read-only and the old code was moved to the legacy branch. So, try this instead to download:
`wget https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code/raw/legacy/Adafruit_DHT_Driver/Adafruit_DHT --output-document Adafruit_DHT`