RPi-Monitor/RPi-Monitor #321: web interface stops responding to connections

jcpvdm, 2020-04-28:
Its the second time it happens, and I'm unable to find the reason. After some unknown number of days, the web interface stops to respond:

tcpdump shows that new TCP connection attempts are ignored
```
sudo tcpdump -n -i any port 8888
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
09:53:23.959015 IP 192.168.1.22.49796 > 192.168.1.148.8888: Flags [S], seq 2528332450, win 64240, options [mss 1460,sackOK,TS val 4210695302 ecr 0,nop,wscale 7], length 0
09:53:24.210113 IP 192.168.1.22.49798 > 192.168.1.148.8888: Flags [S], seq 336953717, win 64240, options [mss 1460,sackOK,TS val 4210695553 ecr 0,nop,wscale 7], length 0
09:53:24.965284 IP 192.168.1.22.49796 > 192.168.1.148.8888: Flags [S], seq 2528332450, win 64240, options [mss 1460,sackOK,TS val 4210696308 ecr 0,nop,wscale 7], length 0
```
ss show stuck entries. They are never expiring. In particular note 192.168.33.6 show ESTAB. This is a connection coming from openvpn client but currently not connected! The CLOSE-WAIT ones also never expire:
```
sudo ss -tap 'dport 8888 or sport 8888'
State               Recv-Q          Send-Q                   Local Address:Port                     Peer Address:Port                                                           
LISTEN              6               5                              0.0.0.0:8888                          0.0.0.0:*               users:(("rpimonitord",pid=6742,fd=4))          
CLOSE-WAIT          2593            0                        192.168.1.148:8888                     192.168.1.22:40856                                                          
CLOSE-WAIT          2625            0                        192.168.1.148:8888                     192.168.1.22:40854                                                          
ESTAB               0               0                        192.168.1.148:8888                     192.168.33.6:60839           users:(("rpimonitord",pid=6742,fd=5))          
CLOSE-WAIT          2646            0                        192.168.1.148:8888                     192.168.1.22:40860                                                          
CLOSE-WAIT          2651            0                        192.168.1.148:8888                     192.168.1.22:40930                                                          
CLOSE-WAIT          2590            0                        192.168.1.148:8888                     192.168.1.22:40858                                                          
CLOSE-WAIT          2550            0                        192.168.1.148:8888                     192.168.1.22:40862       
```

strace on the process 6742 show no activity:
```
sudo strace -p 6742
strace: Process 6742 attached
read(5, 
```

I have never faced this problem with other TCP applications.

apeelme, 2020-04-29:
This is also the case with me

Leepic, 2020-05-03:
Can you tell me if it can be related to #218 ?

jcpvdm, 2020-05-03:
> Can you tell me if it can be related to #218 ?

From time to time I also got the message `"TCP: request_sock_TCP: Possible SYN flooding on port 8888. Sending cookies.  Check SNMP counters."` , but is not correlated to this particular issue. (not always, at least).

Leepic, 2020-05-03:
Can you follow [this comment](https://github.com/XavierBerger/RPi-Monitor/issues/218#issuecomment-450996189) and see if this appens again or not in a few days?

jcpvdm, 2020-05-22:
It's happening again, and there was no syn flooding error on the log. I guess its a different issue.
```
pi@raspberrypi:~ $ ss -tap 'sport 8888'
State             Recv-Q        Send-Q               Local Address:Port                Peer Address:Port
LISTEN            6             5                          0.0.0.0:8888                     0.0.0.0:*
CLOSE-WAIT        489           0                     192.168.33.1:8888                192.168.33.6:52925
CLOSE-WAIT        1             0                     192.168.33.1:8888                192.168.33.6:56677
ESTAB             0             0                    192.168.1.148:8888               192.168.33.10:47444
CLOSE-WAIT        489           0                     192.168.33.1:8888                192.168.33.6:52919
CLOSE-WAIT        516           0                     192.168.33.1:8888                192.168.33.6:56678
CLOSE-WAIT        489           0                     192.168.33.1:8888                192.168.33.6:52974
ESTAB             0             0                    192.168.1.148:8888               192.168.33.10:47445
```

Piradoxlanieve, 2020-06-16:
I have the same problem, the only way it starts up again is by restarting the service, rpimonitor

pchristod, 2020-06-17:
I'm also having this issue. Came here to see if I could find someone who has more insight. Weird thing is that started quite recently, before it would run for months without issues.

telemaxx, 2021-01-02:
0 2 * * *  /usr/sbin/service rpimonitor restart