RPi-Monitor/RPi-Monitor #322: Addons url start with root web server.

chantonoi, 2020-05-03:
When I set reverse proxy form http://127.0.0.1:8888 to https://myhost.com/rpimonitor.
Most data  has work properly, but addons eg. top3 do not work. A top3 addons request at https://myhost.com/addons/top3.html instead of https://myhost.com/rpimonitor/addons/top3.htnl

chantonoi, 2020-05-03:
Okay, it's done, I'm sorry to open this case. It can be easily fixed by edit the cpu.conf. Add '.' Before '/ addons'.

closed 2020-05-03 by chantonoi

Leepic, 2020-05-03:
It seem to be the same issue fixed in #241 (in develop branch).
Can you test the patch explain in [this comment](https://github.com/XavierBerger/RPi-Monitor/issues/241#issuecomment-452902311)?

PS: I sugger you to put an authentication on your public, [as describe in the doc](https://xavierberger.github.io/RPi-Monitor-docs/34_autentication.html#reverse-proxy-configuration), because currently your page is public ;).