RPi-Monitor/RPi-Monitor #326: Cannot find releases using apt

dawez, 2020-05-28:
It seems that there are no more releases in the location specified in the doc:

This is coming from apt update
```
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease                                                                                                                                                                                 
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]                                                                                                                                                                    
Ign:3 http://giteduberger.fr rpimonitor/ InRelease                                                                                                                                                                                            
Err:4 http://giteduberger.fr rpimonitor/ Release                                                                                                                                                                                              
  Redirection loop encountered                                                                                                                                                                                                                
Reading package lists... Done                                                                                                                                                                                                                 
E: The repository 'http://giteduberger.fr rpimonitor/ Release' does no longer have a Release file.                                                                                                                                            
N: Updating from such a repository can't be done securely, and is therefore disabled by default.                                                                                                                                              
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

It seems that the error comes from this url:
```
https://raw.githubusercontent.com/XavierBerger/RPi-Monitor/master/src/etc/apt/sources.list.d/rpimonitor.list
```

Which contains:
```
# RPi-Monitor official repository
deb http://giteduberger.fr rpimonitor/
```

steefje, 2020-05-29:
I had also this problem, this solves the problem: 
edited /etc/apt/sources.list.d/rpimonitor.list and changed the following line from

deb http://giteduberger.fr rpimonitor/

to

deb https://www.giteduberger.fr rpimonitor/

No more errors

cinetube, 2020-06-08:
> 
> 
> I had also this problem, this solves the problem:
> edited /etc/apt/sources.list.d/rpimonitor.list and changed the following line from
> 
> deb http://giteduberger.fr rpimonitor/
> 
> to
> 
> deb https://www.giteduberger.fr rpimonitor/
> 
> No more errors

Thanks for your suggestion. I was also having redirection loop issue. Chaning to deb https://www.giteduberger.fr rpimonitor/" solved the issue. Thank you very much.

Deses, 2021-05-15:
https no longer works.

wtt604, 2022-09-07:
make sure you have the www. in the website, not just https://gitduberger

> https no longer works.

XavierBerger, 2026-10-27:
Installation process has been redesigned.
See README.md for details.

closed 2026-10-27 by XavierBerger