RPi-Monitor/RPi-Monitor #327: Bitdefender shows Trojan Warning in Firefox Profile Cache at loading RPI Monitor Website

Peacemaker80, 2020-05-28:
Hallo,

i don't know  if it is a bug but every time i load or reload the Webinterface of RPIMonitor on my PI3 or PI4 with the newest Raspbian and all updates my Bitdefender Internet Security on my Win 10 Machine comes up with the Warning Message that JS:Trojan.Cryxos.3793 was found in Firefox Profile Cache and move this in Quarantine. After setup a fresh Rasbian install its the same. Anybody else have this issue?
My Win 10 Machine is clean, after full scan and it only appears if i load the RPI Interface.

Thanks for an answer

licaon-kter, 2020-05-28:
But what script? File? Dig into Bitdefender logs

Peacemaker80, 2020-05-28:
C:\Users\XXXX\AppData\Local\Mozilla\Firefox\Profiles\YYYYYYY.default-12345678\cache2\entries\1B946087BA0A8A01FF56A44ACE2EB09D86BBCA68 ist mit JS:Trojan.Cryxos.3793 infiziert.

X, Y and 12345678 are replacements for personal data.
It looks like the cached Data of the Webinterface brings up the Warning. False Positve?

Peacemaker80, 2020-05-28:
I think the file
http://123.123.123.3:8888/js/raphael.2.1.0.min.js
produces the error

Peacemaker80, 2020-05-28:
In Chrome no bitdefender warnings were shown at loading the webinterface of RPI Monitor.
?

licaon-kter, 2020-05-28:
This file? https://github.com/XavierBerger/RPi-Monitor/blob/develop/src/usr/share/rpimonitor/web/js/raphael.2.1.0.min.js

Looks fine: https://www.virustotal.com/gui/file/27b96ab9ad424b0a8088ed06946853db481d1581669b49fc4f4fab2968bd9b36/detection

Can you download the whole repo and scan it locally?