RPi-Monitor/RPi-Monitor #332: Wrong value displayed for the CPU temperature

gilarelli, 2020-06-29:
Hello and first thanks for this great tool.

On my machine, it seems that the CPU temperature displayed on the graph does not reflect the value of the parsed file.
It's almost always more than the value of the file with sometimes a lower value:

[image]
[image]
[image]

I thought at first that it could be some sort of latency, but my CPU doesn't seem to reach some temperatures displayed.

Here is the relevant part of my .conf file:

```dynamic.5.name=soc_temp
dynamic.5.source=/sys/devices/virtual/thermal/thermal_zone0/temp
dynamic.5.regexp=(.*)
dynamic.5.postprocess=sprintf("%.1f", $1/1000)
dynamic.5.rrd=GAUGE

web.status.1.content.1.line.1=JustGageBar("CPU Temperature", "°C", 40, data.soc_temp, 80, 100, 80)
```

I've tried other sources (`/sys/class/thermal/thermal_zone0/temp` and `/etc/armbianmonitor/datasources/soctemp`)with the same result.

Any idea on what could cause this behavior ?

Leepic, 2020-08-02:
Hello,

Maybe it's because RPi-Monitor poll data every 10 seconds (daemon.delay in /etc/rpimonitor/daemon.conf). Is the output with this command match the one in your dashboard?
`which -n10 cat /sys/devices/virtual/thermal/thermal_zone0/temp`

gilarelli, 2020-08-05:
Hi and thanks for your input.

Maybe you're right, but I can't seem to trap the moment when the value is captured as in this screenshot:
[image]

closed 2022-05-18 by gilarelli